raspberrypi/rpi-sb-provisioner #18: Using an HSM instead of file for CUSTOMER_KEY_FILE_PEM?

torntrousers, 2024-08-12:
Hello, this new rpi-sb-provisioner looks super cool!

I wondered if there is any way planned to use an HSM for the signing key instead of just having it in a file ?

torntrousers, 2024-08-12:
Ok!

We have some hacked about code that we did to [raspberrypi/usbboot](https://github.com/raspberrypi/usbboot) for this so that you can use a custom openssl config and pass in a PKCS11 engine sign url like `openssl dgst -engine pkcs11 -keyform engine -sign "pkcs11:object=<keypair-alias>;type=private" -sha256 -out <signed output file> <file_to_sign>`.

So something similar with rpi-sb-provisioner would be great.

Would be happy to work on a PR or at least test out whatever you might do.

torntrousers, 2024-08-13:
How about a PR along the lines of these two main changes:

1) its useful to be able to specify an openssl config file.  This is independent of using PKCS11 HSM and could be useful either way, but its very useful when using PKCS11 so you can have a custom config file to configure the HSM.

So have a new optional configuration field:

```
**OPENSSL_CONFIG_FILE**
**Optional**

The fully qualified path to a custom OPENSSL configuration file

```
Then update everywhere there is an openssl command to use that config file if set:
https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/key-writer/keywriter.sh#L35
https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/host-support/terminal-functions.sh#L8
https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/device-provisioner/provisioner.sh#L214
https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/device-provisioner/provisioner.sh#L380

2) Update to support an pkcs11 engine key url instead of a key file

```
**OPENSSL_USE_PKCS11**
**Optional**

Use a key from a PKCS11 HSM instead of a file. 
When using PKCS11 the CUSTOMER_KEY_FILE_PEM should be set with a value for the HSM key, eg: "pkcs11:object=<keypair-alias>;type=private"   

Set to any value to enable using a PKCS11 HSM. 

```
Then update everywhere there is an openssl command to change "-keyform PEM" to be "-engine pkcs11 -keyform engine" 

Eg from:

`${OPENSSL} dgst -sign "${CUSTOMER_KEY_FILE_PEM}" -keyform PEM -sha256 "${TMP_DIR}"/boot.img`

to:

`${OPENSSL} dgst -sign "${CUSTOMER_KEY_FILE_PEM}" -engine pkcs11 -keyform engine -sha256 "${TMP_DIR}"/boot.img`

Note: this is slightly imperfect as its reusing the CUSTOMER_KEY_FILE_PEM configuration field to not be file path but contain the PKCS11 key name. Maybe you'd prefer a new configuration field or this? (though that does make it more complicated)

torntrousers, 2024-08-13:
Ok, maybe in that case it doesn't really need the OPENSSL_USE_PKCS11 configuration parameter and instead could have: 

---

**CUSTOMER_PKCS11_KEY**
**Optional**

Define the PKCS#11 key value when the openssl configuration supports using a PKCS#11 engine. It should be set with a value for the HSM key, for example: "pkcs11:object=<keypair-alias>;type=private"   

When CUSTOMER_PKCS11_KEY is set then the CUSTOMER_KEY_FILE_PEM parameter will be ignored.

---

Also, presently CUSTOME_KEY_FILE_PEM is defined as *Mandatory* and in [triage.sh](https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/device-triage/triage.sh#L23) it polices that, so those would need to be updated too.

tdewey-rpi, 2024-08-13:
Suggestion accepted, @torntrousers 

I'm looking to get the outstanding PRs merged today, but once those look to be progressed I'll move on to adding HSM support per this issue.

PR #20 (2024-08-16): Add support for HSM-based signing
Add rudimentary support for using a PKCS11 engine to provide the private key for the signed boot component.

This does not impact the device-unique key that is generated as part of provisioning.

Where the PKCS11_NAME and FILE_PEM are both supplied, favour the PKCS11 name.

Fixes #18
closed 2024-08-16 by tdewey-rpi