raspberrypi/rpi-sb-provisioner #19: a device unique key?

torntrousers, 2024-08-12:
Hello again.

I the README about the provisioner phase it says it creates a device unique key but furkliny around i can't find any other mention about that key or what its for or where its kept. Is it the LUKSv2 encryption key perhaps? Where is the code that creates it? Could there maybe be a little more in the README about it?

Thanks for your comments!

torntrousers, 2024-08-12:
Thanks for the quick and detailed reply.

Thats as descibed in the Device-specific private key section of this doc https://pip.raspberrypi.com/categories/685-whitepapers-app-notes/documents/RP-003611-WP/Using-the-One-time-programmable-memory-on-Raspberry-Pi-single-board-computers.pdf?

Ok, i'll wait for the opening up the fastboot-based support image.

Thanks again.

tdewey-rpi, 2024-08-12:
Yes, @torntrousers, that's the record we're using.

closed 2024-08-13 by torntrousers