raspberrypi/rpi-sb-provisioner #26: Textual dependencies difficult to meet

roliver-rpi, 2024-08-20:
`config.sh' and `monitor.sh' require a version of python3-textual, python3-real that is not available in Debian Bookworm.  Whilst Debain packages that rpi-sb-provisioner requirements are available in Trixie, this would require modification of package sources / priorities (or manual package download/installation) which is less-than-desirable.

Investigate the [dh-virtualenv](https://github.com/spotify/dh-virtualenv) project (or similar) to determine whether a venv-in-a-package approach offers a more user-friendly solution.

tdewey-rpi, 2024-08-21:
I believe the only sensible path forward is to change our packaging mechanism from `nfpm` to `debuild` - we're now requiring more complicated features that warrant bringing in the additional complexity of `debuild`.

tdewey-rpi, 2024-08-23:
This should now be covered in the next release - the replacement build system will construct a venv as part of the Debian package, and that will inherit the correct Textual dependencies.

tdewey-rpi, 2024-08-29:
Fixed in v1.0.2