raspberrypi/rpi-sb-provisioner #28: Newbie questions

torntrousers, 2024-08-21:
Hello, i'm trying to use this to provision a device and have some questions:

I can't get the TUI commands to work,  config.sh and monitor.sh. I see there is an open issue about the Textual dependencies but is there any way to use these commands in the meantime?

It's not so clear what happens when DEMO_MODE_ONLY is set - does any image at all get written to the CM4 or can you only do the triage phase? 

The triage phase seems to work ok and the triage log ends with "Keywriting completed. Rebooting for next phase." and then nothing happens. Is that supposed to be an automatic reboot or must I manually power cycle the CM4? Thats what i end up doing, should the the disable eMMC Boot pins jumper be connected for this or not? 

Eventually power cycling without the jumper on the provisioner phase seems to run and the end of the log shows this:

```
 .  .  .
        Starting boot.img signing
+ echo ================================================================================
================================================================================
+ sha256sum /tmp/tmp.oFyz8IEzfo/boot.img
+ awk '{print $1}'
+ echo -n 'rsa2048: '
+ openssl dgst -sign /home/ant/private.pem -keyform PEM -sha256 /tmp/tmp.oFyz8IEzfo/boot.img
+ xxd -c 4096 -p
+ announce_stop 'boot.img signing'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Stopping boot.img signing'

        Stopping boot.img signing
+ echo ================================================================================
================================================================================
+ announce_start 'Boot Image partition extraction'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Starting Boot Image partition extraction'

        Starting Boot Image partition extraction
+ echo ================================================================================
================================================================================
++ stat -c%s /tmp/tmp.oFyz8IEzfo/boot.img
+ REQUIRED_BOOTIMG_SIZE=48234496
++ stat -c%s /tmp/tmp.oFyz8IEzfo/boot.sig
+ REQUIRED_BOOTSIG_SIZE=587
++ stat -c%s /tmp/tmp.oFyz8IEzfo/config.txt
+ REQUIRED_CONFIGTXT_SIZE=106
+ SECTOR_SIZE=512
+ TOTAL_SIZE=48235189
+ TOTAL_SIZE=48235253
+ TOTAL_SIZE=47105
+ SECTORS=92
+ SECTORS=46
+ dd if=/dev/zero of=/tmp/tmp.oFyz8IEzfo/bootfs-temporary.img bs=1M count=512
512+0 records in
512+0 records out
536870912 bytes (537 MB, 512 MiB) copied, 0.455972 s, 1.2 GB/s
+ mkfs.fat -n BOOT /tmp/tmp.oFyz8IEzfo/bootfs-temporary.img
mkfs.fat 4.2 (2021-01-31)
++ mktemp -d
+ META_BOOTIMG_MOUNT_PATH=/tmp/tmp.ynlkR5NlVU
+ mount -o loop /tmp/tmp.oFyz8IEzfo/bootfs-temporary.img /tmp/tmp.ynlkR5NlVU
+ cp /tmp/tmp.oFyz8IEzfo/boot.img /tmp/tmp.ynlkR5NlVU/boot.img
+ cp /tmp/tmp.oFyz8IEzfo/boot.sig /tmp/tmp.ynlkR5NlVU/boot.sig
+ cp /tmp/tmp.oFyz8IEzfo/config.txt /tmp/tmp.ynlkR5NlVU/config.txt
+ sync
+ sync
+ sync
+ umount /tmp/tmp.ynlkR5NlVU
+ rm -rf /tmp/tmp.ynlkR5NlVU
+ mv /tmp/tmp.oFyz8IEzfo/bootfs-temporary.img /srv/pi-sb-provisioner.XjP/bootfs-temporary.img
+ announce_stop 'Boot Image partition extraction'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Stopping Boot Image partition extraction'

        Stopping Boot Image partition extraction
+ echo ================================================================================
================================================================================
+ announce_start 'Erase / Partition Device Storage'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Starting Erase / Partition Device Storage'

        Starting Erase / Partition Device Storage
+ echo ================================================================================
================================================================================
+ '[' -z 1 ']'
+ sleep 2
+ '[' -z 1 ']'
+ sleep 2
+ '[' -z 1 ']'
+ sleep 2
+ '[' -z 1 ']'
+ sleep 2
+ '[' -z 1 ']'
+ sleep 2
+ '[' -z 1 ']'
+ sleep 2
+ announce_stop 'Erase / Partition Device Storage'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Stopping Erase / Partition Device Storage'

        Stopping Erase / Partition Device Storage
+ echo ================================================================================
================================================================================
+ announce_start 'Resizing OS images'
+ echo ================================================================================
================================================================================
+ printf '\n\t%s\n' 'Starting Resizing OS images'

        Starting Resizing OS images
+ echo ================================================================================
================================================================================
++ get_variable partition-size:mapper/cryptroot
++ fastboot getvar partition-size:mapper/cryptroot
++ grep -oP 'partition-size:mapper/cryptroot: \K.*'

```
and thats it and nothing else happens. not clear if its finished ok or not? Eventually rebooting the CM4 with a HDMI screen attached shows this:

[image]

so it doesn't seem to have made the eMMC bootable?

In the end i'd like to be able to test out the HSM support from https://github.com/raspberrypi/rpi-sb-provisioner/pull/20#issuecomment-2293521114 but thats not yet out in a release. Do i need to somehow make my own build from the latest code?

tdewey-rpi, 2024-08-21:
> Hello, i'm trying to use this to provision a device and have some questions:
> 
> I can't get the TUI commands to work,  config.sh and monitor.sh. I see there is an open issue about the Textual dependencies but is there any way to use these commands in the meantime?

I'll have to defer to @bebon901 for that - a venv is almost certainly the right path forward, but if you're willing to live (slightly) dangerously, you could install the Debian Trixie versions of Textual. We were caught out by the lack of updates to Textual in the Debian project - Bookworm ships with a horrendously old version, which is where you'll be seeing problems.

> 
> It's not so clear what happens when DEMO_MODE_ONLY is set - does any image at all get written to the CM4 or can you only do the triage phase?

No images are written to CM4 at all - but your device runs through the entire state machine, generating a keywriter payload, a fastboot payload, and the modified partitions. I introduced the flag to allow developers iterating on the process (or indeed, on their base OS) to inspect intermediates ahead of writes. As the writes can take a while (they're very much size dependant), this was a concrete productivity win.

> 
> The triage phase seems to work ok and the keywriter log ends with "Keywriting completed. Rebooting for next phase." and then nothing happens. Is that supposed to be an automatic reboot or must I manually power cycle the CM4? Thats what i end up doing, should the the disable eMMC Boot pins jumper be connected for this or not?

This is the expected flow when DEMO_MODE_ONLY is set - as no image is _actually_ written, the automatic reboot commands are not transferred. You shouldn't need to disconnect the jumper pins at any time during provisioning.

> Eventually power cycling without the jumper on the provisioner phase seems to run and the end of the log shows this:
>
> ```
>  .  .  .
>
>         Starting boot.img signing
> + echo [...]

This looks like fastboot has failed to open your cryptroot. That's a missing DEMO_MODE_ONLY wrapper, so that's definitely something we should fix. I can see you had DEMO_MODE_ONLY set from these lines:

> + '[' -z 1 ']'
> + sleep 2

> and thats it and nothing else happens. not clear if its finished ok or not? Eventually rebooting the CM4 with a HDMI screen attached shows this:
> 
> ![cm4boot](https://private-user-images.githubusercontent.com/12925578/359766031-366eaf2f-ce79-428f-88e3-900c91c7d2b6.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQyMzMxMTQsIm5iZiI6MTcyNDIzMjgxNCwicGF0aCI6Ii8xMjkyNTU3OC8zNTk3NjYwMzEtMzY2ZWFmMmYtY2U3OS00MjhmLTg4ZTMtOTAwYzkxYzdkMmI2LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIxVDA5MzMzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZjBjOWQ1ZTE4ODQ4MThjMWRkODZmMjYwN2QzZjRjYmZhMmUyNGUyNWVhNGRlYjhjNzI5YjFlY2RiYzM2ZTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.pes_EFLsC-3_GQ1Nq4EaXahV0R6jAzdlmRkxLFKgIec)
> 
> so it doesn't seem to have made the eMMC bootable?

Yes - if you've only set provisioner to run in DEMO_MODE, then it will _not_ result in a bootable eMMC.

> 
> In the end i'd like to be able to test out the HSM support from [#20 (comment)](https://github.com/raspberrypi/rpi-sb-provisioner/pull/20#issuecomment-2293521114) but thats not yet out in a release. Do i need to somehow make my own build from the latest code, and if so how would i do that?

At the simplest level:

`npfm package -p deb` && `dpkg -I $package_name.deb`

I am, however, about to cut a release for the HSM initial functionality, so you might be better served by waiting.

tdewey-rpi, 2024-08-29:
v1.0.2 is now released - including the monitoring and configuration apps. DEMO_MODE_ONLY fix will move into v1.0.3, due next week.

torntrousers, 2024-08-29:
That great , we will give it a try

tdewey-rpi, 2024-09-13:
@torntrousers I believe v1.1.0 will now have resolved your outstanding questions - let me know how you get on.

torntrousers, 2024-09-13:
thanks, will go give it a try ..

torntrousers, 2024-10-03:
One of my colleagues was the one who tried the 1.1.0 release and rasied https://github.com/raspberrypi/rpi-sb-provisioner/issues/45 which i see is now closed. Bit unclear to me from looking tat the issue tracker if things are all good to try now or should i wait for the 1.2.0 release, and if so what sort of time frame might that be?
(many thanks!)

tdewey-rpi, 2024-10-03:
I'd strongly encourage waiting for 1.2.0 - we're addressing a major concern in the pre-boot authentication image, and while HEAD is in a much better shape I'm still reticent to suggest moving forward without the updated cryptroot.

I'm expecting to cut that release either this week or early next week (possibly even breaking my Don't Deploy On Fridays! rule).

torntrousers, 2024-10-08:
with the 1.1.1 release we've i think successfully setup a CM4 with secure boot. yay, thank you. Is there as easy way to see it really is? lsblk shows the root partition is using crypto_LUKS, i can see there is a boot.sig in the /boot/firmware directory. Is there some command that shows the cm4 is using really secure boot or the efuse is correctly set or something, couldn't see anything in dmesg?

tdewey-rpi, 2024-10-08:
The best way to determine if secure boot is being enforced by the bootloader is probably to have the payload check.

There's a bit of documentation (https://github.com/raspberrypi/documentation/blob/a4826cbba4190ffb376026a5e7726c5c9d3fe9ee/documentation/asciidoc/computers/configuration/device-tree.adoc?plain=1#L931) around a field in the /chosen/bootloader device tree file, which gets populated by the system.

Note that you can also use that to perform a safety check in a deployed image - if you ever see that _not_ being true, abort execution/fire flares/call home.

tdewey-rpi, 2024-10-08:
Secondly, you can ask rpiboot to fetch device metadata - which will include a hash of the customer OTP, under the 'CUSTOMER_KEY_HASH' field of the resultant JSON.

If you ask rpi-sb-provisioner to provision your device again - and do so with RPI_DEVICE_FETCH_METADATA configured - you'll get the metadata as part of the provisioning flow in `/var/log/rpi-sb-provisioner/<serial>/metadata/<serial>.json`

torntrousers, 2024-10-08:
> The best way to determine if secure boot is being enforced by the bootloader is probably to have the payload check.
>
> There's a bit of documentation [...]

```
hexdump /proc/device-tree/chosen/bootloader/signed
0000000 0080 0900
```

Could that 8 mean "The customer public key digest has been written to OTP", but i'm not sure its in the correct place?

tdewey-rpi, 2024-10-08:
@torntrousers Assuming you're happy with the explanations so far, I'm going to close this issue so we can fracture out new questions into individual issues.

closed 2024-10-08 by tdewey-rpi