raspberrypi/rpi-sb-provisioner #42: Provisioning fastboot Error

Siecje, 2024-09-13:
```
	Starting Erase / Partition Device Storage
+ echo ================================================================================
================================================================================
+ '[' -z '' ']'
+ fastboot erase mmcblk0
< waiting for any device >
Erasing 'mmcblk0'                                  OKAY [  0.006s]
Finished. Total time: 0.020s
+ sleep 2
+ '[' -z '' ']'
+ fastboot oem partinit mmcblk0 DOS ''
FAILED (remote: 'Invalid id. e.g:
	oem partinit mmcblk0 DOS [id]')
fastboot: error: Command failed
+ cleanup
+ mkdir -p /var/log/rpi-sb-provisioner/a0f4790c/
+ echo PROVISIONER-EXITED
+ unmount_image ''
+ sync
+ sleep 1
++ losetup --list
++ grep ''
++ cut -f1 '-d '
+ LOOP_DEVICE=
+ '[' -n '' ']'
+ '[' -d /tmp/tmp.nnHJWpzXYq ']'
+ rm -rf /tmp/tmp.nnHJWpzXYq
+ '[' -f '' ']'

```

tdewey-rpi, 2024-09-13:
Paging @roliver-rpi for visibility

LimaLima, 2024-09-22:
clearing "RPI_SB_WORKDIR" worked as a workaround for me.
too bad the "successful" provisioning still doesn't result in a bootable cm4 with a default "lite" image :(

tdewey-rpi, 2024-09-24:
@Siecje Did you get this log from the monitor script? I've just found an interesting bug where the provisioner might have stale logs.

Siecje, 2024-09-24:
It was from the provisioning log file.

tdewey-rpi, 2024-10-04:
I'd be really keen to see what you get with v1.1.1, and if you still see this, @Siecje.

Thanks,

Tom

closed 2024-10-07 by Siecje