raspberrypi/rpi-sb-provisioner #70: How to disable rpi-sb-provisioner

Siecje, 2024-10-09:
After installing Version 1.2.0 and rebooting I don't seem to have a service running

```shell
systemctl list-units rpi-sb-provisioner*
```

```
  UNIT LOAD ACTIVE SUB DESCRIPTION
0 loaded units listed. Pass --all to see loaded but inactive units, too.
To show all installed unit files use 'systemctl list-unit-files'.
```

I would like to stop the service so that I can manually run rpiboot.

Siecje, 2024-10-09:
I do have a .service file but the service is not found.

```shell
$ cd /usr/local/lib/systemd/system 
/usr/local/lib/systemd/system $ cat rpi-sb-provisioner@.service 
[Unit]
Description=Provision a Raspberry Pi Device with a authentication key, a device-unique key, and an OS that you supply.

[Service]
Type=oneshot
ExecStart=/usr/local/bin/rpi-sb-provisioner.sh %I
EnvironmentFile=/etc/rpi-sb-provisioner/config
/usr/local/lib/systemd/system $ sudo service rpi-sb-provisioner status
Unit rpi-sb-provisioner.service could not be found.
```

rpi-sb-provisioner is installed.
```shell
$ dpkg -s rpi-sb-provisioner
Package: rpi-sb-provisioner
Status: install ok installed
Priority: optional
Section: admin
Installed-Size: 116479
Maintainer: Tom Dewey <tom.dewey@raspberrypi.com>
Architecture: arm64
Version: 1.2.0
Depends: fastboot (>= 33.0.3), python3, python3-pycryptodome, openssl, cpio, sed, android-sdk-platform-tools, awk, xxd, rpi-eeprom (>= 26), rpiboot, coreutils, curl, bash, gzip, dctrl-tools, diffutils, findutils, libengine-pkcs11-openssl, libp11-kit-dev, gnutls-bin, jq
Pre-Depends: dpkg (>= 1.16.1)
Conffiles:
 /etc/default/rpi-sb-provisioner 27158014ce808e1f35615456a9d2c4b7
Description: Automated provisioning of secure boot for Raspberry Pi Devices
Homepage: https://www.raspberrypi.com/software

```

tdewey-rpi, 2024-10-10:
I'll add this to the docs - but essentially, your safest move is to just disable the udev trigger, in the manner described in this AskUbuntu thread: https://askubuntu.com/questions/240632/how-to-disable-udev-net-rule-generation

I'd go for something like...

`sudo ln -s /dev/null /etc/udev/rules.d/60-rpi-sb-provisioner.rules`

Don't forget to delete that symlink when you want to turn it back on. It will persist across reboots.

PR #72 (2024-10-10): README: How to disable rpi-sb-provisioner
Fixes: #70
closed 2024-10-10 by tdewey-rpi