raspberrypi/rpi-sb-provisioner #81: Provisioning results in a root Busybox shell

JasonJibrail, 2024-11-19:
I have followed the readme to provision a CM4 with a pi 5

Generated an image with PiGen - standard 64 bit Lite bookworm - boots fine when programmed with imager

Provisioning succeeds ok, but when I reboot the device I see a busybox shell that I can log in to as 'root' with no pass 
Looks like a ramdisk.

Have I missed a step to end up here?

JasonJibrail, 2024-11-19:
[rpi-sb-provisioner.zip](https://github.com/user-attachments/files/17819160/rpi-sb-provisioner.zip)
Attached are logs for two serials (with metadata) and the config file

There's nothing product specific in here, it's a vanilla pios for me to test the provisioner and the keys used will be discarded.

tdewey-rpi, 2024-11-20:
Excellent, thanks for providing those @JasonJibrail 

Logs all look OK - it appears the OS was modified and written correctly. The prompt you see is indeed the pre-boot authentication environment, and it looks like you should have had all the key material available to perform the unlock & pivot.

At this point, it'd be useful to confirm the version of rpi-sb-provisioner you're using:

```apt info rpi-sb-provisioner```

Finally, obtaining the `dmesg` from one of the devices would be extremely helpful.

Finally, it's worth noting that the devices are absolutely in a recoverable position - but we may need to send them through the provisioner again to ensure all items are as expected. We shouldn't do that before confirming why the pre-boot authentication environment failed to open the storage, though.

If you would prefer, you can send this information to the new disclosure e-mail group, which is only accessible by rpi-sb-provisioner maintainers that are Raspberry Pi staff. Your disclosure would only be used in the context of an issue report - and should reference this GitHub issue. `sb-provisioner-private @ raspberrypi.com`

JasonJibrail, 2024-11-21:
I've replied via the new disclosure email, thanks

tdewey-rpi, 2024-11-22:
Issue identified and resolved in ae74aee39343f045dfd1d41644ab1e9f8b3e2a97.

Essentially, the pre-boot authentication image was manually inserting kernel modules a hold-over from a previous build system. With the new pi-gen-micro based image, we can rely on automatic insertion - which came in to a clash with the older way of handling the crypto modules.

closed 2024-11-22 by tdewey-rpi

T-droneops, 2024-12-01:
Hi, I have the same issue with v.1.3.0. Freshly installed Raspberry Pi os, Rpi 5 provisioning CM4 with a pi-gen generated lite image.

T-droneops, 2024-12-01:
The behavior is quite weird. I have one .img made by pi-gen roughly a week ago, and it works well with the sb-provisioner. But when I try to make a new image, it always fails to boot, remaining stuck in bysybox. Also there seems to be a problem with retrieving keypair from the device. Provisioning log always stops at this stage, but in the keyfile directory I see only one empty file $CM$_SERIAL.der. Also monitor.sh doesn't show successfully finished devices, the list just becomes empty after some time. The OS, provisioner and pi-gen were all installed today on a new SD card.

tdewey-rpi, 2024-12-02:
@T-droneops Could you please make a new issue to track the pi-gen/sb-provisioner mismatch you're seeing.

In terms of the device keypair - the current coded behaviour is to fail-safe if you're asking for the private key from a device that has already been provisioned - but I'll release a v1.3.1 that fleshes out this failure further and allows you to continue.

T-droneops, 2024-12-02:
@tdewey-rpi , I've created one. Thank you for the quick response.

roliver-rpi, 2024-12-03:
I'm able to reproduce this issue, looking at the contents of `/cryptroot_stderr.log`, we get:
```
Cannot initialize device-mapper. Is dm_mod kernel module loaded?
Cannot use device cryptroot, name is invalid or still in use.
```

Looking in /lib/modules I have `6.6.51+rpt-rpi-v8`, and `6.6.62+rpt-rpi-{2712,v8}`; the 6.6.62 v8 kernel is currently running. Unfortunately, only the 6.6.51 kernel has the generated `/lib/modules/6.6.51+rpt-rpi-v8/modules.dep` (and other associated files) required for auto-loading of modules via modprobe.

I will update augment_initramfs() (https://github.com/raspberrypi/rpi-sb-provisioner/blob/8868a383a54305db5b4d2deb0840926f8bb5f609/service/rpi-sb-provisioner.sh#L771) to generate depmod information.

T-droneops, 2024-12-04:
Hello,

Thank you for quick check of this issue. Is there any workaround I can do
while waiting for the fixed release? I'm using sb-provisioner in production
and now it's blocked.
Currently I am doing next: I provision the module with my old. img, it
succeeds in provisioning, but this OS version is too old for me. After the
provisioning, I mount encrypted /rootfs as a volume on my laptop and
manually copy /rootfs from the new OS image. It works but I don't think
it's a proper way to do provisioning.

Best regards,
T-Droneops

roliver-rpi, 2024-12-04:
Hello @T-droneops , https://github.com/raspberrypi/rpi-sb-provisioner/pull/87 addresses this issue.