raspberrypi/rpi-sb-provisioner #94: Install - dpkg

tennisparty, 2024-12-09:
Hello, 

I am trying to install the software via dpkg as in the docs but I am getting an error that the system cannot find the *.deb file. Is there an alternative method for installing?

I am running on a fresh image of the raspberry pi 5 bookworm 64bit on a raspberry pi 5

```pi@5book:~ $ sudo apt update && sudo apt full-upgrade -y
Hit:1 http://deb.debian.org/debian bookworm InRelease
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:3 http://deb.debian.org/debian-security bookworm-security InRelease
Hit:4 http://deb.debian.org/debian bookworm-updates InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  xinput
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@5book:~ $ sudo dpkg -i rpi-sb-provisioner_foo.deb
dpkg: error: cannot access archive 'rpi-sb-provisioner_foo.deb': No such file or directory```

roliver-rpi, 2024-12-11:
Did you download the latest version from the releases area on GitHub?
https://github.com/raspberrypi/rpi-sb-provisioner/releases/tag/v1.3.0

You need to download: rpi-sb-provisioner_1.3.0_arm64.deb and then, assuming the file is in your downloads folder: `sudo dpkg -i ~/Downloads/rpi-sb-provisioner_1.3.0_arm64.deb`

AndriiNeivanov, 2024-12-13:
Same issue. :) It is not obvious. Please add it into the README.

tdewey-rpi, 2024-12-17:
We've gone slightly further than that - and added it to the Raspberry Pi APT repositories.

`sudo apt install -y rpi-sb-provisioner`