raspberrypi/rpi-sb-provisioner #97: Loading Fastboot failed, timed out.

AndriiNeivanov, 2024-12-17:
Provisioning failed with `Loading Fastboot failed, timed out.` message.

Steps to reproduce:
Make provisioning as described in official documentation. RPI 5 should make provisioning of CM 4.

Logs:
[provisioner.log](https://github.com/user-attachments/files/18162807/provisioner.log)
[keywriter.log](https://github.com/user-attachments/files/18162813/keywriter.log)
[triage.log](https://github.com/user-attachments/files/18162815/triage.log)
[progress.log](https://github.com/user-attachments/files/18162825/progress.log)

After failure not able to use `rpiboot`:
```
$ sudo ./rpiboot 
[sudo] password for pc: 
RPIBOOT: build-date Dec 17 2024 version 20240422~085300 62f37dae
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.
Waiting for BCM2835/6/7/2711/2712...
Loading embedded: bootcode4.bin
Sending bootcode.bin
Successful read 4 bytes 
Waiting for BCM2835/6/7/2711/2712...
Loading embedded: bootcode4.bin
Second stage boot server
Cannot open file boot.sig
Cannot open file boot.sig
Cannot open file boot.sig
Cannot open file boot.sig
Cannot open file boot.sig
```

tdewey-rpi, 2024-12-17:
Thanks for the report, @AndriiNeivanov

Inspecting the logs, it appears that you've written the key successfully - which explains future `rpiboot` strangeness if you don't also sign those components.

What is unexpected, however, is Fastboot timing out.

With the device in such a state, are you able to get the result from `fastboot devices`? Note that this output will contain a serial number - which I strongly recommend you obfuscate (replace it with X's or similar).

AndriiNeivanov, 2024-12-17:
Probably it was a power loss issue. It looks like CM 4 has power loss during provisioning.
So, I moved forward and hang on `Starting Resizing OS images` stage in `provisioning.log`:
[provisioner.log](https://github.com/user-attachments/files/18166105/provisioner.log)
[keywriter.log](https://github.com/user-attachments/files/18166106/keywriter.log)
[progress.log](https://github.com/user-attachments/files/18166107/progress.log)
[triage.log](https://github.com/user-attachments/files/18166108/triage.log)

Only red led is glowing.

Monitor lost the device:
[image]

>With the device in such a state, are you able to get the result from fastboot devices? Note that this output will contain a serial number - which I strongly recommend you obfuscate (replace it with X's or similar).

It looks like `fastboot` works well now:
```
$ fastboot devices
xxxxxxxxxxxxxxxx	fastbootd
```

tdewey-rpi, 2024-12-17:
Thanks for the update, @AndriiNeivanov.

One thing you can try is 'forgetting' the device entirely.

Delete the entire log set for that device:

```rm -rf /var/log/rpi-sb-provisioner/xxxxxxx/```

And then re-plugging the power of the compute module IO board.

AndriiNeivanov, 2024-12-18:
>One thing you can try is 'forgetting' the device entirely.
Delete the entire log set for that device:
`rm -rf /var/log/rpi-sb-provisioner/xxxxxxx/`
And then re-plugging the power of the compute module IO board.

I deleted and tried again, and had failure in `provisioner.log`:
`Loading Fastboot failed, timed out.`
[progress.log](https://github.com/user-attachments/files/18175931/progress.log)
[provisioner.log](https://github.com/user-attachments/files/18175932/provisioner.log)
[keywriter.log](https://github.com/user-attachments/files/18175933/keywriter.log)
[triage.log](https://github.com/user-attachments/files/18175934/triage.log)

Also the `monitor.sh` had failure.

tdewey-rpi, 2024-12-18:
Thanks for confirming @AndriiNeivanov 

@roliver-rpi will you have chance to look at this?

roliver-rpi, 2024-12-18:
@timg236 , are you familiar with this RPIBOOT failure mode? Unfortunately, any error returned by the 'rpiboot' executable has been dropped as rpi-sb-provisioner.sh only checks for a timeout error when loading the fastboot gadget.

oliora, 2024-12-19:
I've got the same issue and stuck now.
Maybe important is that I had to change `SOURCE_EEPROM_IMAGE` from `"/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-15.bin"` because the original file does not exist.

oliora, 2024-12-19:
Have restarted the process and now I'm getting keywriter error
```
Writing key and EEPROM configuration to the device
Writing failed, timed out.
```

AndriiNeivanov, 2024-12-20:
> I've got the same issue and stuck now. Maybe important is that I had to change `SOURCE_EEPROM_IMAGE` to `"/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-07.bin"` because the original eeprom file does not exist.

The same on my RPI 5 (prov server) for CM 4 (prov client).

```
$ ls -la bootloader-2712/latest 
total 8304
drwxr-xr-x 2 root root    4096 Dec 19 14:59 .
drwxr-xr-x 4 root root    4096 Dec 19 14:59 ..
-rw-r--r-- 1 root root 2097152 Dec 16 11:13 pieeprom-2024-11-12.bin
-rw-r--r-- 1 root root 2097152 Dec 16 11:13 pieeprom-2024-11-27.bin
-rw-r--r-- 1 root root 2097152 Dec 16 11:13 pieeprom-2024-12-07.bin
-rw-r--r-- 1 root root 2097152 Dec 16 11:13 pieeprom-2024-12-15.bin
-rw-r--r-- 1 root root  102804 Dec 16 11:13 recovery.bin

$ ls -la bootloader-2711/latest 
total 6348
drwxr-xr-x 2 root root   4096 Dec 19 14:59 .
drwxr-xr-x 4 root root   4096 Dec 19 14:59 ..
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2023-01-11.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2023-05-11.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-01-18.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-01-22.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-04-15.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-04-17.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-05-17.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-07-30.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-09-05.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-10-10.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-10-21.bin
-rw-r--r-- 1 root root 524288 Dec 16 11:13 pieeprom-2024-12-07.bin
-rw-r--r-- 1 root root  97604 Dec 16 11:13 recovery.bin
-rw-r--r-- 1 root root  99352 Dec 16 11:13 vl805-000138c0.bin
```

I fixed path on latest and it resolved the problem. From now on my CM 4 has success boot.

@roliver-rpi, @tdewey-rpi, please add the patch below to main branch and to the Raspberry Pi APT repositories.
```
git diff dev/eeprom-image-fix..main
index f6a6f5e..de31d8d 100755
--- a/service/rpi-sb-provisioner.sh
+++ b/service/rpi-sb-provisioner.sh
@@ -370,12 +370,12 @@ BOOTCODE_BINARY_IMAGE=
 BOOTCODE_FLASHING_NAME=
 case ${RPI_DEVICE_FAMILY} in
     4)
-        SOURCE_EEPROM_IMAGE="/lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2024-12-07.bin"
+        SOURCE_EEPROM_IMAGE="/lib/firmware/raspberrypi/bootloader-2711/latest/pieeprom-2024-09-05.bin"
         BOOTCODE_BINARY_IMAGE="/lib/firmware/raspberrypi/bootloader-2711/latest/recovery.bin"
         BOOTCODE_FLASHING_NAME="${FLASHING_DIR}/bootcode4.bin"
         ;;
     5)
-        SOURCE_EEPROM_IMAGE="/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-15.bin"
+        SOURCE_EEPROM_IMAGE="/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-09-23.bin"
         BOOTCODE_BINARY_IMAGE="/lib/firmware/raspberrypi/bootloader-2712/latest/recovery.bin"
         BOOTCODE_FLASHING_NAME="${FLASHING_DIR}/bootcode5.bin"
         ;;
```

roliver-rpi, 2024-12-20:
The SOURCE_EEPROM_IMAGE certainly needs updating for CM5 as the original file no longer exists as @oliora points out.

I'm still not sure why you've been facing issues on CM4 @AndriiNeivanov as bootloader-2711/latest/pieeprom-2024-09-05.bin is still available; I've been able to provision a CM4 using rpi-sb-provisioner 1.3.1 installed from the apt repository without issue.

AndriiNeivanov, 2024-12-20:
It looks like my CM 4 does not work with `pieeprom-2024-09-05.bin`, but work well with `pieeprom-2024-12-07.bin`.

oliora, 2024-12-20:
Even with the recent EEEROM file the rpiboot times out for me, effectively never hearing back from RPi 5. The log is the following:

```
RPIBOOT: build-date Nov  7 2024 version 20241107~183156
Loading: /tmp/tmp.Sn0ZqN5hRV/bootcode5.bin
Waiting for BCM2835/6/7/2711/2712...
```

tdewey-rpi, 2025-02-14:
1.3.6 includes a fix to calculate the eeprom to use based on the newest release from the rpi-eeprom package.

I believe all sub-issues raised in this issue have been resolved, so I'm going to mark it as 'Fixed'. Please raise a new issue if you encounter new problems.

closed 2025-02-14 by tdewey-rpi