raspberrypi/rpi-sb-provisioner #99: Documentation is not clear regarding how Raspberry Pi should be connected to be provisioned

oliora, 2024-12-18:
The readme file contains picture with Raspberry Pi connected to Compute Module but no picture how to connect a Raspberry Pi for provisioning.

tdewey-rpi, 2025-01-06:
@oliora Are you asking for expanded documentation on:

* Provisioning a Raspberry Pi form-factor device; or
* Setting up the Raspberry Pi as the provisioning system

oliora, 2025-01-06:
The former: provisioning RPi5 from RPi5

tdewey-rpi, 2025-01-06:
Understood.

Provisioning a Raspberry Pi 5 requires a manual intervention for each device. A CM5 can be put into RPIBOOT mode with a jumper wire - but on Raspberry Pi 5, you need to do this by holding the power button as you insert the power & data cable. Make sure your SD card is already inserted, and you have no other peripherals connected.

Beyond that, however, the provisioning flow behaves identically to the CM5 flow.

oliora, 2025-01-07:
@tdewey-rpi thank you for the details. Any recommendations regarding the most reliable and convenient way to power the RPi5 during provisioning?

tdewey-rpi, 2025-01-07:
I'd strongly recommend using a powered hub - the Raspberry Pi USB hub would be a fair choice, combined with a Raspberry Pi 5 power supply.

That said, for provisioning with just an SD card attached, I've been able to power from the server Pi5 supplied by the Raspberry Pi 5 power supply

oliora, 2025-01-07:
Do you mean this one? https://thepihut.com/products/raspberry-pi-usb-3-hub

For me provisioning with only official 27W USB power adapter did not work: the provisioned Raspberry with SD card was hanging at the fastboot stage but there is a small chance that my power adapter is defective.

tdewey-rpi, 2025-01-08:
> Do you mean this one? https://thepihut.com/products/raspberry-pi-usb-3-hub
> 
> For me provisioning with only official 27W USB power adapter did not work: the provisioned Raspberry with SD card was hanging at the fastboot stage but there is a small chance that my power adapter is defective.

ACK. That's not something I've encountered in my test rigs at all.

Can you confirm:

- What's connected physically to the provisioning server
- What's connected physically to the provisioning client set-up
- Any details of the USB-A to USB-C cable you're using to connect the server to the client.

A picture would be ideal - but if your opsec forbids that, text would be great.

oliora, 2025-02-01:
@tdewey-rpi Sorry, was tight on time. I've checked my setup and strangely I was not using the 27W power adapter but the official 15W one which explains my issue with power supply. I'll check the process with 27W adapter and be back

oliora, 2025-02-03:
Works like a charm with the official 27W power adapter

tdewey-rpi, 2025-02-03:
Glad to hear you've found success, @oliora.

I'll consider some wording to make clear the dependency on the 27W supply for the Pi5 => [Pi5]n case.

Sravdar, 2025-03-24:
Hello i am quite new to this subject. Could you, @tdewey-rpi,  clarify if it is possible to provision Raspbbery Pi4 Model B as well? If yes how so since it doesn't have a power button.

tdewey-rpi, 2025-05-19:
Closing as 'fixed', per comments on Feb 3rd. Docs now explicitly request Raspberry Pi 27W power supply.

closed 2025-05-19 by tdewey-rpi