raspberrypi/rpi-sb-provisioner #101: deactivating LUKS disk encryption in config file?

jpiwek, 2024-12-21:
Is there a chance to deactivate LUKS encryption feature?
It would be a good option for testing and debugging.

tdewey-rpi, 2025-01-06:
Thanks for the question, @jpiwek 

re: LUKS
This isn't something I've entertained previously, because I'm not entirely clear how it would assist testing without creating a conspicuous provisioning foot-gun.

Because the LUKSv2 container creation and OS insertion is handled automatically, and the intermediate images are already capable of being analysed offline, I can't see what use case this LUKS-for-all model blocks - could you walk me through the requirement?

re: New partitions
Yes, you'd want to use the `partapp` command. Use the example of the command that creates the boot partition to see how to create known-size partitions. If you do not specify a size, `partapp` will assume you want to fill the remainder of the storage media.

See:
```
[ -z "${DEMO_MODE_ONLY}" ] && fastboot oem partapp "${RPI_DEVICE_STORAGE_TYPE}" 0c "$(stat -c%s "${RPI_SB_WORKDIR}"/bootfs-temporary.img)"
```

Where 0c is the fdisk partition code, and the last argument is calculating the size of the partition required.

re: Documentation of `partapp`

We're intending to open-source our fastboot changes this month, which should give you complete visibility - but for now, the description above should help you move forward.

tdewey-rpi, 2025-01-07:
> re: LUKS one use-case is legacy unencrypted images without LUKSv2

Unencrypted images are ingested by rpi-sb-provisioner, which can apply LUKSv2 and perform the necessary modifications without repeated user intervention. LUKSv2 might cost some throughput - but I wouldn't expect a legacy image to pose a problem.

> another use-case I play with is LUKSv2 encryption with TPM2

I'm not sure what you'd gain from that without using signed boot and some careful mitigations for bus sniffing, but this is absolutely something you could explore by generating a new cryptroot image (see pi-gen-micro for that)

jpiwek, 2025-01-09:
@tdewey-rpi 
> Unencrypted images are ingested by rpi-sb-provisioner, which can apply LUKSv2 and perform the necessary modifications without repeated user intervention. LUKSv2 might cost some throughput - but I [...]
re: What I meant is make a configuration option available to bypass LUKSv2 container creation. perhaps the configuration can be made in genimage.cfg.in

> > another use-case I play with is LUKSv2 encryption with TPM2
> 
> I'm not sure what you'd gain from that without using signed boot and some careful mitigations for bus sniffing, but this is absolutely something you could explore by generating a new cryptroot image (see pi-gen-micro for that)

re: Rpi internal signed boot should be enabled. I did some investigation on using TPM2 for 2 use-cases:

- U-Boot: compiled U-Boot with TPM2 driver support to measure specific payload.... 
- LUKSv2:  there is no secure storage on Rpi and vcmailbox with LUKS decryption key can practically being read by everyone. systemd-cryptenroll with TPM2 might extend security level. the TPM pub key is stored in the LUKS header which is used to establish a encrypted connection between host and TPM. the decryption key is then transfered encrypted to prevent bus sniffing attacks.

jpiwek, 2025-01-10:
@timg236: Thanks for your interest

I will try to answer your question and pull more details together... at the moment few things are under NDA. What is the best way to get in contact (email,...?) with you to share preliminary details as long it's not fully public yet? I have an NDA with Raspberry Pi in place.

I am talking to a TPM2 semiconductor supplier and have a call next week. 
Furthermore I am looking into creating 2 low cost TPM2 boards to demonstrate these things and will then go public.

timg236, 2025-01-10:
@tdewey-rpi might be the best contact here - btw I don't believe u-boot works with secure-boot

jpiwek, 2025-01-10:
@timg236 : I did a try and build U-Boot with TPM2 support.... the approach should be as follow 

High Level Approach I have in mind, few steps already work with my CM4 board (with headless Bookworm image) and TPM2:
- First Stage Rpi Secure Boot loads signed boot.img  
- in config.txt: kernel=u-boot.bin option needs to be changed to load U-Boot binary script
- TPM2 support in u-boot available. PCRs could be used to measure payloads (e.g. config files, Kernel, Initramfs,...) 
- a simple U-Boot boot script loads the kernel8.img -> should be adapted to load initramfs + kernel  payload

let me know your thoughts.

tdewey-rpi, 2025-01-10:
> @timg236 : I did a try and build U-Boot with TPM2 support.... the approach should be as follow
>
> High Level Approach I have in mind, few steps already work with my CM4 board (with headless Bookworm [...]

I suspect you could greatly simplify this flow, and eliminate the TPM2 from your design:

* Raspberry Pi Bootloader Loads bespoke cryptroot, containing per-OTA measurement data for config files, initramfs, downstream kernels
* Bespoke cryptroot uses measurement data to authenticate future stages
* Bespoke cryptroot calls kexec() and pivot_root() to migrate to the 'next' stage.

But in practice, because the Cryptroot is a signed item (and something you have absolute dominion over), you could even avoid the kexec - and then just OTA your cryptroot _and_ your downstream userspace as one atomic unit.

The primary advantage I'd see from the TPM2 is if you wanted to protect against key recovery after your OS is compromised - but since the only recoverable key would be device-specific, the attack would have extremely limited value.

The fundamental concern I have with introducing U-Boot is one of risk management - U-Boot is a complex project containing a broad range of functions and designs, that in nearly every professional setting I've seen it used in has been misconfigured or misdeployed. The fewer components in the TCB, the smaller the attack surface, and the greater confidence developers can have in producing great products.

jpiwek, 2025-01-10:
@tdewey-rpi : do you talk just about the provisioning process or general approach to protect keys during operation of the device?
my goal is to protect keys with TPM2 to unlock LUKS paritions during boot and to provide secure key storage during RPi4/5 device operation to verify fimware updates, license management,..... The TPM2 has great support in Linux and it is a generic solution approach.

If there is a way to reduce complexity by getting rid off of U-Boot from my approach and directly integrate the TPM2 into Rpi initramfs to measure integratity of payloads, then let's have an offline discussion about option and benefits. @timg236 had few ideas and question: 

>  if this is used as a one-shot authentication to a TPM then we can make this a one-shot API so only the first caller after power-on will get the key.
> re you able to share any information about where the measured payload is stored and which parts of the system use it e.g. kernel / user space?
> 
> The Pi4/Pi5 signed boot system already validate the contents of boot.img (kernel, firmware, configuration data, initramfs) against the root of trust. If the measured data is located inside of the LUKS system (e.g. downloaded) then a kernel TPM driver can be used. In principle, the sha256 of boot.img can be made available to the OS via device tree.

Please send me your email address. 
I would like to discuss few things offline. 

@timg236 :
this is how TPM2 PCR can be used with systemd-cryptenroll:
https://manpages.debian.org/bookworm/systemd/systemd-cryptenroll.1.en.html

Thanks

tdewey-rpi, 2025-01-13:
> @tdewey-rpi : do you talk just about the provisioning process or general approach to protect keys during operation of the device? my goal is to protect keys with TPM2 to unlock LUKS paritions during [...]

I deal in general platform security, among other things - so any recommendation comes from the view of the full device lifecycle, from initial prototyping through to eventual disposal or RMA.

We've probably reached the point of this discussion where we should migrate to the disclosure e-mail list. This e-mail list is visible to a restricted subset of Raspberry Pi employees who interact with the bootloader, the provisoner, and general platform security.

Please contact sb-provisioner-private atsymbol raspberrypi.com, referencing this issue for context.

Closing to avoid inadvertent disclosures of design details.