raspberrypi/rpi-sb-provisioner #104: Unexpected error while using rpi-sb-provisioner

EWagRP, 2025-01-08:
Hello there,

i´m trying to get used to secure boot on raspberry pi 5.
Since I know about the rpi-sb-provisioner, i was, really excited to try it out and provide a nice and secure solution concerning signed and encrypted boot processes.
I followed the instructions on https://github.com/raspberrypi/rpi-sb-provisioner to get everything done right.
Unfortunately openssl seems to fail at some point and I have no clue how to fix it.

Can anybody out there please help me getting this process done?

pem file was generated this way: 
`openssl genrsa 2048 > privatekey.pem`

here is my config: /etc/rpi-sb-provisioner/config 
`CUSTOMER_KEY_FILE_PEM=/home/pi/privatekey.pem`
`CUSTOMER_KEY_PKCS11_NAME="pkcs11:object=<keypair-alias>;type=private"`
`GOLD_MASTER_OS_FILE=/home/pi/2025-01-02-raspios-bookworm-armhf.img`
`RPI_DEVICE_STORAGE_TYPE=sd`
`RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64`
`RPI_DEVICE_FAMILY=5`
`RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config`
`RPI_DEVICE_LOCK_JTAG=`
`RPI_DEVICE_EEPROM_WP_SET=`
`RPI_DEVICE_FETCH_METADATA=`
`RPI_DEVICE_RETRIEVE_KEYPAIR=`
`DEMO_MODE_ONLY=`
`RPI_SB_WORKDIR=`
`RPI_SB_PROVISONER_MANUFACTURING_DB=`

rpi-sb-provisioner.sh #OUTPUT: 
`+ TARGET_DEVICE_SERIAL=`
`+ DEBUG=`
`+ OPENSSL=openssl`
`+ export KEYWRITER_FINISHED=KEYWRITER-FINISHED`
`+ export KEYWRITER_ABORTED=KEYWRITER-ABORTED`
`+ export KEYWRITER_STARTED=KEYWRITER-STARTED`
`+ export PROVISIONER_FINISHED=PROVISIONER-FINISHED`
`+ export PROVISIONER_ABORTED=PROVISIONER-ABORTED`
`+ export PROVISIONER_STARTED=PROVISIONER-STARTED`
`+ : aes-xts-plain64`
`+ echo KEYWRITER-STARTED`
`+ read_config`
`+ [ -f /etc/rpi-sb-provisioner/config ]`
`+ . /etc/rpi-sb-provisioner/config`
`+ CUSTOMER_KEY_FILE_PEM=/home/pi/privatekey.pem`
`+ CUSTOMER_KEY_PKCS11_NAME=pkcs11:object=<keypair-alias>;type=private`
`+ GOLD_MASTER_OS_FILE=/home/pi/2025-01-02-raspios-bookworm-armhf.img`
`+ RPI_DEVICE_STORAGE_TYPE=sd`
`+ RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64`
`+ RPI_DEVICE_FAMILY=5`
`+ RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config`
`+ RPI_DEVICE_LOCK_JTAG=`
`+ RPI_DEVICE_EEPROM_WP_SET=`
`+ RPI_DEVICE_FETCH_METADATA=`
`+ RPI_DEVICE_RETRIEVE_KEYPAIR=`
`+ DEMO_MODE_ONLY=`
`+ RPI_SB_WORKDIR=`
`+ RPI_SB_PROVISONER_MANUFACTURING_DB=`
`+ CUSTOMER_PUBLIC_KEY_FILE=`
`+ TMP_DIR=`
`+ trap cleanup EXIT`
`+ mktemp -d`
`+ FLASHING_DIR=/tmp/tmp.zBhXHvinG0`
`+ derivePublicKey`
`+ mktemp`
`+ CUSTOMER_PUBLIC_KEY_FILE=/tmp/tmp.Dad5UlClaP`
`+ openssl rsa -in /home/pi/privatekey.pem -pubout`
`writing RSA key`
`+ identifyBootloaderConfig`
`+ [ ! -f /var/lib/rpi-sb-provisioner/bootloader.config ]`
`+ enforceSecureBootloaderConfig`
`+ grep -Fxq SIGNED_BOOT=1 /var/lib/rpi-sb-provisioner/bootloader.config`
`+ echo boot_ramdisk=1`
`+ echo uart_2ndstage=1`
`+ SOURCE_EEPROM_IMAGE=`
`+ DESTINATION_EEPROM_IMAGE=`
`+ DESTINATION_EEPROM_SIGNATURE=`
`+ BOOTCODE_BINARY_IMAGE=`
`+ BOOTCODE_FLASHING_NAME=`
`+ SOURCE_EEPROM_IMAGE=/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-19.bin`
`+ BOOTCODE_BINARY_IMAGE=/lib/firmware/raspberrypi/bootloader-2712/latest/recovery.bin`
`+ BOOTCODE_FLASHING_NAME=/tmp/tmp.zBhXHvinG0/bootcode5.bin`
`+ DESTINATION_EEPROM_IMAGE=/tmp/tmp.zBhXHvinG0/pieeprom.bin`
`+ DESTINATION_EEPROM_SIGNATURE=/tmp/tmp.zBhXHvinG0/pieeprom.sig`
`+ [ ! -e /tmp/tmp.zBhXHvinG0/pieeprom.sig ]`
`+ [ ! -e /lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-19.bin ]`
`+ update_eeprom /lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-19.bin /tmp/tmp.zBhXHvinG0/pieeprom.bin home/pi/privatekey.pem /tmp/tmp.Dad5UlClaP`
`+ src_image=/lib/firmware/raspberrypi/bootloader-2712/latest/pieeprom-2024-12-19.bin`
`+ dst_image=/tmp/tmp.zBhXHvinG0/pieeprom.bin`
`+ pem_file=/home/pi/privatekey.pem`
`+ public_pem_file=/tmp/tmp.Dad5UlClaP`
`+ sign_args=`
`+ [ -n /home/pi/privatekey.pem ]`
`+ grep -q SIGNED_BOOT=1 /var/lib/rpi-sb-provisioner/bootloader.config`
`+ mktemp`
`+ TMP_CONFIG_SIG=/tmp/tmp.Pa2omVmBv8`
`+ keywriter_log Signing bootloader config`
`+ echo Signing bootloader config`
`+ writeSig /var/lib/rpi-sb-provisioner/bootloader.config /tmp/tmp.Pa2omVmBv8`
`+ mktemp`
`+ SIG_TMP=/tmp/tmp.L2eeFJKucb`
`+ IMAGE=/var/lib/rpi-sb-provisioner/bootloader.config`
`+ OUTPUT=/tmp/tmp.Pa2omVmBv8`
`+ sha256sum /var/lib/rpi-sb-provisioner/bootloader.config`
`+ awk {print $1}`
`+ date -u +%s`
`+ echo ts: 1736237614`
`+ get_signing_directives`
`+ [ -n pkcs11:object=<keypair-alias>;type=private ]`
`+ echo pkcs11:object=<keypair-alias>;type=private -engine pkcs11 -keyform engine`
`+ [ -n pkcs11:object=<keypair-alias>;type=private -engine pkcs11 -keyform engine ]`
`+ get_signing_directives`
`+ [ -n pkcs11:object=<keypair-alias>;type=private ]`
`+ echo pkcs11:object=<keypair-alias>;type=private -engine pkcs11 -keyform engine`
`+ openssl dgst -sign pkcs11:object=<keypair-alias>;type=private -engine pkcs11 -keyform engine -sha256 -out /tmp/tmp.L2eeFJKucb /var/lib/rpi-sb-provisioner/bootloader.config`
`Engine "pkcs11" set.`
`Failed to enumerate slots`
`PKCS11_get_private_key returned NULL`
`Could not read private key from org.openssl.engine:pkcs11:pkcs11:object=<keypair-alias>;type=private`
`40C0B259FF7F0000:error:40800067:pkcs11 engine:ERR_ENG_error:invalid parameter:eng_back.c:603:`
`40C0B259FF7F0000:error:13000080:engine routines:ENGINE_load_private_key:failed loading private key:../crypto/engine/eng_pkey.c:79:`
`+ cleanup`
`+ [ -d  ]`
`+ rm /tmp/tmp.Dad5UlClaP`
`+ unmount_image `
`+ sync`
`+ sleep 1`
`+ losetup --list`
`+ grep `
`+ cut -f1 -d `
`+ LOOP_DEVICE=`
`+ [ -n  ]`
`+ [ -d  ]`
`+ [ -f  ]`
`+ [ -n  ]`

all I can say is that the error appears in the writeSig() function in rpi-sb-provisioner.sh,
during this command: `"${OPENSSL}" dgst -sign $(get_signing_directives) -sha256 -out "${SIG_TMP}" "${IMAGE}"`

i really wonder why i run into these issues:
`Failed to enumerate slots`
`PKCS11_get_private_key returned NULL`
`Could not read private key from org.openssl.engine:pkcs11:pkcs11:object=<keypair-alias>;type=private`

If someone can suppurt me, please let me know.

tdewey-rpi, 2025-01-08:
Thanks for the report, @EWagRP.

Your issue is in your config file - you've set both PKCS11_NAME and FILE_PEM, and when both are set rpi-sb-provisioner will prefer the PKCS11 name over the file-backed key.

If your private key is available on disk, use FILE_PEM and delete the value for PKCS11_NAME. Otherwise, if you're using a HSM or similar to store your key, make sure your PKCS11 object name is correct - `<keypair_alias>` looks like a value taken from a tutorial, rather than your actual PKCS11 key object name.

EWagRP, 2025-01-08:
@tdewey-rpi that´s it!

You helped me so much!! 
I´m so greatful for that. 
And I keep with the PEM solutuion and deleted the PKCS11 setting.

closed 2025-01-08 by tdewey-rpi