raspberrypi/rpi-sb-provisioner #106: Clear failed devices

reflexcameras, 2025-01-17:
Hello,

I have tried using the sb-provisioner and it didn't work on two CM4 devices that I connected. I have tried reconnecting the devices but they do not process through the monitor anymore. I deleted the files at /var/log/rpi-sb-provisioner/ and the custom workdir to try and reset their variables for re-provisioning but they don't seem to be detected. 

The images written to the CM4 were not changed.

Is there a way to reset the provisioner to re-detect those devices?

Many thanks,
Sam

tdewey-rpi, 2025-01-17:
Definitely a monitor bug, thanks for the report Sam.

The most straightforward way to reset the monitor is to re-start it.

reflexcameras, 2025-01-17:
In the end uninstalled and re-installed the software and that worked on a refresh. Is there a way to re-provision the same device but with different settings?

reflexcameras, 2025-01-18:
I managed to reflash by deleting the directory listed at the bottom of the docs and the workdir. I didn't have any issue so all sorted.

closed 2025-01-18 by reflexcameras

reflexcameras, 2025-01-22:
Hi, what's the best way to restart the monitor service. I can't see a systemd service for the monitor script in the code