raspberrypi/rpi-sb-provisioner #107: Trigger script after provision

reflexcameras, 2025-01-18:
Is there a way I could automate triggering a script after each provision? I would like to edit the hostname and add a linked deploy key to the image I am using. I am making the deploy keys on the provisioning device

reflexcameras, 2025-01-21:
Great, thankyou, I will try this and make sure I don't set the WORKDIR so that it regenerates on each provision.

If I installed the sb-provisioner via apt where will I find the rpi-sb-provisioner.sh to edit?

tdewey-rpi, 2025-01-22:
Installed from `apt`, you'll find it in `/usr/local/bin`, as specified in our `install` file:

https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/debian/install

reflexcameras, 2025-01-22:
amazing, thank you

tdewey-rpi, 2025-04-29:
Good news! I've now added a customisation hook and a script editor to the 2.0 WebUI to allow you to do this in a more straightforward manner. There might even be some relevant help text.

Closing as 'fixed', separate issues for problems with the customisation hooks and script editor please.

closed 2025-04-29 by tdewey-rpi