raspberrypi/rpi-sb-provisioner #111: Edit /boot/firmware/config.txt after provisioning

reflexcameras, 2025-01-28:
I would like to swap cameras on a provisioned device. I cannot find config.txt in the usual directory /boot/firmware/config.txt. Is there a way to edit this file after provisioning to change the overlays?

tdewey-rpi, 2025-01-28:
`config.txt` behaves differently in the secure-boot universe.

In the non-secure boot mode, it's simply read from the VFAT as part of the boot sequence - but when you migrate to secure-boot, you're instead embedding it in the Pre-Boot Authentication image.

You essentially have two paths forward:

1. (Recommended) Regenerate the cryptroot Pre-boot authentication initramfs with your required changes. This will let you prune the overlays to just those you require, and ensure the overlays you require are present. Use pi-gen-micro to do this: https://github.com/raspberrypi/pi-gen-micro/tree/main/configurations/cryptroot_image
2. (Quick, but potentially with missing files) Add your directives to your installation: https://github.com/raspberrypi/rpi-sb-provisioner/blob/d377e82357fa9acebee96381dd538ed12a87bbe4/service/rpi-sb-provisioner.sh#L870

Note also that (2) will _not_ survive an upgrade of the provisioner - and we're planning to drop 2.0 within the next few weeks with a range of improvements and changes.

reflexcameras, 2025-01-28:
OK, thank you, is step 1 possible from the provisioned device or just the provisioner? It would be amazing for testing (and future proofing) if it is possible to edit the overlays on a provisioned device.

Still, I want to create a golden image for my project. Would the config files update if I:
- remove my previous wkdir
- generate a new pi-gen image with a corrected config.txt 
- reflash the same device

Sorry for the basic questions.

tdewey-rpi, 2025-01-28:
> OK, thank you, is step 1 possible from the provisioned device or just the provisioner? It would be amazing for testing (and future proofing) if it is possible to edit the overlays on a provisioned device.

It is indeed possible - just replace the boot.img & boot.sig with your updated copies. Provided you sign them with the same signing key, this should work well.

> 
> Still, I want to create a golden image for my project. Would the config files update if I:
> 
> * remove my previous wkdir
> * generate a new pi-gen image with a corrected config.txt
> * reflash the same device

No - you'd need to update the pre-boot authentication image for overlays, and this would be a separate input to sb-provisioner.

To do so, make your modifications in pi-gen-micro, and build a new initramfs there, and then replace the following file with your newly minted cryptroot:

https://github.com/raspberrypi/rpi-sb-provisioner/blob/d377e82357fa9acebee96381dd538ed12a87bbe4/debian/install#L25

> 
> Sorry for the basic questions.

tdewey-rpi, 2025-01-29:
@reflexcameras I must apologise, because I think I've sent you down a wrong path.

Inspecting the code once more, I can confirm we _only_ use the initramfs from the pi-gen-micro output, and this would not include config.txt.

Instead, it _should_ be as straightforward as modifying your `gold master` image, and making sure your requisite overlay files are present in your boot image. You should be able to confirm this without going through a secure boot provisioning run - simply inspecting your `gold master` should be enough to confirm presence.

These changes will then be fully captured in the boot.img. I had forgotten that the cryptroot is kept as a subitem in the boot.img - alongside the config.txt, cmdline.txt and traditional boot flow configuration items.

See https://github.com/raspberrypi/rpi-sb-provisioner/blob/d377e82357fa9acebee96381dd538ed12a87bbe4/service/rpi-sb-provisioner.sh#L898 for verification

reflexcameras, 2025-01-29:
No worries, thanks for confirming. Now that I have changed and reflashed the gold master image, it is working! 

Thanks again,
Sam

PR #134 (2025-02-14): README: Expand logging, config.txt
Expand the logging section to include the runes for real-time observation of a provisioning flow.

Additionally, add a section on changing config.txt.

Fixes #111
commit 95873b61 (2025-02-14): README: Expand logging, config.txt

Expand the logging section to include the runes for real-time observation of a provisioning flow.

Additionally, add a section on changing config.txt.

Fixes #111
closed 2025-02-14 by tdewey-rpi