raspberrypi/rpi-sb-provisioner #121: Secure boot provisioning over network boot

tcbennun, 2025-02-06:
Hi folks. Apologies if this is regarded as off-topic, but here goes.

The [cmprovision](https://github.com/raspberrypi/cmprovision) tool is a really nifty thing – mass concurrent CM4 provisioning with nothing but a network cable; no temporary jumpers required. If your host board has PoE, then you don't even need a power lead!

That all being said, it appears to be kind of abandoned, and there's no secure boot support. If I want SB, looks like the best option is this repository, but having to set up USB + power + temporary jumper is not ideal if you really wanna optimise the production line (and at the volumes we're looking at, we do!!)

Do you folks see any reason why SB provisioning over network boot can't be done? In my simple world, it seems like a custom buildroot image would be booted over network, which deploys the EEPROM config before turning into the fastboot gadget - which has TCP support, right? The remainder of the steps in `rpi-sb-provisioner.sh` are just fastboot commands, so... am I missing something?

It would be a really, really cool thing. :shrug:

Suggestions and advice very welcome! Thanks in advance!

tdewey-rpi, 2025-02-06:
> Do you folks see any reason why SB provisioning over network boot can't be done?

You cannot set up secure boot by any mechanism _other_ than USB - so to do this with cmprovision you'd need to create a multi-stage provisioner inside a single-stage provisioner.

Faced with that constraint, I concluded the only sensible path forward was to design a new provisioner, and avoid introducing risk to existing cmprovision users.

Practically, rpi-sb-provisioner can fill the niche of cmprovision in a straightforward way, using simpler constructs and code.

In terms of a production line, a fairly straightforward process change gets you to where you want to be:

1) Set up a provisioning server. I'd recommend Raspberry Pi 5 with an NVME drive, or similar.
2) Install rpi-sb-provisioner, configure that accordingly. Run the monitor either locally with a HDMI display, or over Raspberry Pi Connect for remote.
3) Set up a compute module IO board, or provisioning jig to act as your flashing station. Connect this to the provisioning server by USB, or by USB & Ethernet after an upcoming PR.
4) With power to the compute module IO board or provisioning jig, insert a Compute Module.
5) Apply power to the Compute module
6) Remove power, and the compute module from the flashing station (to be inserted into your final plastics/specialised IO board)
7) GOTO 4

Circling back to the initial question - no, there is no plan to implement writing secure boot configuration from Linux userspace. The scope for abuse is far too great compared to the process changes individual entities would have to make to use rpi-sb-provisioner.

closed 2025-02-06 by tdewey-rpi

tcbennun, 2025-02-06:
Thanks for the quick reply @tdewey-rpi .

I was under the impression EEPROM was writeable from userspace but I'm glad to be corrected.

Final question if that's ok: does `rpi-sb-provisioner` handle concurrent provisions? The service seems to be started on a per-serial-number basis.

tdewey-rpi, 2025-02-06:
> Thanks for the quick reply [@tdewey-rpi](https://github.com/tdewey-rpi) .
> 
> I was under the impression EEPROM was writeable from userspace but I'm glad to be corrected.

While the EEPROM can be written from userspace, there are additional actions that require the use of RPIBOOT - the usbboot repo contains more information in this area, if you're curious.

> 
> Final question if that's ok: does `rpi-sb-provisioner` handle concurrent provisions? The service seems to be started on a per-serial-number basis.

Yes - the system is designed to have a common crossbar service (triage) that starts in response to a UDEV add of an RPIBOOT device, which then starts the main provisioner service. Each provisioner service is designed to operate independently - however you may want to flash one device first, and then expand to concurrent (to ensure your modified-flashing-image is correct).

tcbennun, 2025-02-06:
That's been really useful @tdewey-rpi , thank you!!

One thing outstanding for me right now: we would love to be able to have the CM4 automatically reboot into our OS once provisioning is complete. I can open a separate issue for this, if it's not a quick answer

tdewey-rpi, 2025-02-06:
I'm currently looking at such a feature - but it'll wind up being hardware-dependent. The software support is there in Fastboot, but you will need to programmatically control the nRPIBOOT GPIO.

tcbennun, 2025-02-06:
Got it, thank you. Our hardware will have a solution for the nRPIBOOT manipulation. If we can use fastboot to trigger GPIOs, we should be all set.