raspberrypi/rpi-sb-provisioner #130: Any way to speed up the rootfs write?

tcbennun, 2025-02-12:
Hi,

I'm using a Raspberry Pi 4B as host, provisioning a CM4 (32 GB eMMC); rpi-sb-provisioner at 1.3.3. Everything pretty much worked out the box (cool!!!)

One issue at the minute. Rootfs write takes some 16 minutes, which might be a blocker for our contract manufacturer. Is this due to having to write the entire full-sized encrypted fs? My OS image is only around 2 GB.

If you know of a route towards optimising this, I'd be interested to know.

Many thanks!

tdewey-rpi, 2025-02-12:
Thanks for the report, @tcbennun 

I suspect the primary issue is the lack of crypto acceleration on CM4 (CM5 gains AES instructions that improve performance dramatically).

The best mitigation is an upcoming PR, which will automatically convert your image into a sparse image - which means we won't have to transfer irrelevant sections.

I'll link the PR as it comes up.

tdewey-rpi, 2025-02-12:
PR merged in #131.

I'm about to pop 1.3.6 out, having run it against a CM5 machine where it's made a serious transfer time improvement (334s -> 227s for a Raspberry Pi OS Desktop image).