raspberrypi/rpi-sb-provisioner #136: Digital signal from provisioned device to indicate provisioning completion

tcbennun, 2025-02-14:
This continues a conversation started [here](https://github.com/raspberrypi/rpi-sb-provisioner/issues/121).

We have a device comprising both a CM4 and an on-board microcontroller. In the manufacturing flow, we will need to perform steps involving the MCU after the CM4 provisioning is complete. This will be achievable if we can output a digital signal from the provisioned CM4 to the MCU.

Now, out of the box we reckon we could mod the board to use `LED_nPWR` for this, given it's used as an indicator to the user that the provisioning is done:

https://github.com/raspberrypi/rpi-sb-provisioner/blob/170825abddbacbc56cf858096b4e5ae316d4e41a/service/rpi-sb-provisioner.sh#L1031

This is a little hacky, though. I don't know whether this pin gets driven during CM4 startup. It would be nice if we could use fastboot to set a digital output on any pin!

Thanks!

tcbennun, 2025-03-07:
Sorry to be forward, but thought I'd bump this now the 2.0 PR is open. @tdewey-rpi 

Would be cool to hear any thoughts on this. If the fastboot gadget source code is public I'd happily implement it myself.

tdewey-rpi, 2025-03-07:
@tcbennun Could you share more details on what variety of signal would work with your board?

Additionally - why would a signal from the provisioning server not be sufficient?