raspberrypi/rpi-sb-provisioner #140: Why no option to set "revoke_devkey=1"

jesperbnp, 2025-03-03:
Hi,

I'm working on enabling secure boot for our product based on cm4.

I feel there should be an option of setting revoke_devkey=1 when programming the OTP. Is there any reason why this is not included?

tdewey-rpi, 2025-03-04:
Thanks for the report, @jesperbnp 

This option wasn't plumbed through because it wasn't marked supported on Raspberry Pi 5-family devices.

Thankfully, the option is fairly trivial to add as a manual step - which is why we implemented this project primarily in shell scripts rather than a binary.

jesperbnp, 2025-03-04:
Ok. Thanks for the info. I'll just patch it into the shell script on my side.

Might be worth mentioning it in the readme as I guess it still applies to rpi4/cm4

closed 2025-03-04 by jesperbnp