raspberrypi/rpi-sb-provisioner #144: FASTBOOT_DEVICE_SPECIFIER is empty if clients are not enumerated in time

jesperbnp, 2025-03-08:
Sorry for spamming issues, but I'm trying to set up a production line using rpi-sb-provisioner and seem to be running into multiple issues :-)

FASTBOOT_DEVICE_SPECIFIER sometimes returns empty from this line. Especially when multiple USB devices has to be enumerated at the same time:
https://github.com/raspberrypi/rpi-sb-provisioner/blob/170825abddbacbc56cf858096b4e5ae316d4e41a/service/rpi-sb-provisioner.sh#L938

I added a crude retry/timeout loop which repeats until the device is detected:

``` bash
echo "Looking for $TARGET_DEVICE_SERIAL" >&2
FASTBOOT_DEVICE_SPECIFIER=""
TIMEOUT=0
set +e
while [ -z "$FASTBOOT_DEVICE_SPECIFIER" ]; do
  sleep 1
  FASTBOOT_DEVICE_SPECIFIER="$(timeout 5 fastboot devices | grep "${TARGET_DEVICE_SERIAL}" | cut -f1)"
  echo "$TIMEOUT Found device: $FASTBOOT_DEVICE_SPECIFIER" >&2
  TIMEOUT=$(( TIMEOUT + 1 ))
  if [ "$TIMEOUT" -ge 30 ]; then
    break
  fi
done
echo "Found device specifier: $FASTBOOT_DEVICE_SPECIFIER" >&2
set -e
```

With above retry mechanism I finally got it working. As can be seenn the devices are finally enumerated about 10 seconds after hitting the code:

``` 
Mar 08 22:40:08 nnsecure bash[94691]: Looking for 6c9bc4e3
Mar 08 22:40:09 nnsecure bash[94691]: 0 Found device:
Mar 08 22:40:11 nnsecure bash[94691]: 1 Found device:
Mar 08 22:40:12 nnsecure bash[94691]: 2 Found device:
Mar 08 22:40:13 nnsecure bash[94691]: 3 Found device:
Mar 08 22:40:14 nnsecure bash[94691]: 4 Found device:
Mar 08 22:40:16 nnsecure bash[94691]: 5 Found device:
Mar 08 22:40:16 nnsecure kernel: usb 1-1.2: new high-speed USB device number 49 using xhci-hcd
Mar 08 22:40:17 nnsecure kernel: usb 1-1.2: device descriptor read/64, error -71
Mar 08 22:40:18 nnsecure kernel: usb 1-1.2: New USB device found, idVendor=18d1, idProduct=4e40, bcdDevice= 1.00
Mar 08 22:40:18 nnsecure kernel: usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 08 22:40:18 nnsecure kernel: usb 1-1.2: Product: Raspberry Pi Compute Module 4 Rev 1.1
Mar 08 22:40:18 nnsecure kernel: usb 1-1.2: Manufacturer: Raspberry Pi
Mar 08 22:40:18 nnsecure kernel: usb 1-1.2: SerialNumber: 100000006c9bc4e3
Mar 08 22:40:18 nnsecure mtp-probe[103075]: checking bus 1, device 49: "/sys/devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2"
Mar 08 22:40:18 nnsecure mtp-probe[103075]: bus: 1, device: 49 was not an MTP device
Mar 08 22:40:18 nnsecure mtp-probe[103076]: checking bus 1, device 49: "/sys/devices/platform/axi/1000120000.pcie/1f00200000.usb/xhci-hcd.0/usb1/1-1/1-1.2"
Mar 08 22:40:18 nnsecure mtp-probe[103076]: bus: 1, device: 49 was not an MTP device
Mar 08 22:40:18 nnsecure bash[94691]: 6 Found device:
Mar 08 22:40:19 nnsecure bash[94691]: 7 Found device: 100000006c9bc4e3
Mar 08 22:40:19 nnsecure bash[94691]: Found device specifier: 100000006c9bc4e3
```

tdewey-rpi, 2025-03-17:
Thanks for the report, @jesperbnp 

I was greatly surprised by this report - mostly because I've never personally observed this behaviour.

2.0 will include a different trigger for Fastboot - one that ensures Fastboot stages only begin _after_ the kernel has enumerated the device. I suspect, therefore, we're about to have designed this problem out.

tdewey-rpi, 2025-04-29:
I believe this has now been designed-out, as we only trigger later stages of the provisioner in response to udev enumerating a fastboot descriptor.

Closing as 'fixed'.

closed 2025-04-29 by tdewey-rpi