raspberrypi/rpi-sb-provisioner #145: rpi-sb-provisioner fails if multiple instances run simultaneously

jesperbnp, 2025-03-08:
Somewhat related to issue #144 ... I'm trying to provision multiple boards at a time.
It seems rpi-sb-provisioner was partially prepared for this but definitely have some quirks and lack of testing. Please correct me if I'm wrong :-)

When multiple instances of rpi-sb-provisioner are started at the same time none of them succeeds.
Easily reproduced by hooking up three boards to the provisioning server and switch all three clients ON simultaneously.

Observed problems:
- fastboot boot.img/boot.sig is regenerted for every instance with same filenames i.e. files are constantly overwritten while being used by other instances.
- bootfs/rootfs image generation is similar issue. But can be mitigated as it is only built once.
- Timeout when getting FASTBOOT_DEVICE_SPECIFIER. I raised that as a seperate issue #144 as it is not directly related to multiple instances.

I got it working reliably on my end now (I think). The quick'n'dirty solution I'm going with for now is to put a lock around most of the code so only one instance gets to run at a time. Except for the actual time-consuming fastboot flash commands which are allowed to run in parallel.

tdewey-rpi, 2025-03-10:
Thanks for the report, @jesperbnp!

We're indeed aware of a race around bootfs/rootfs generation. I didn't consider the fastboot re-generation to be an issue - but as you've highlighted that would indeed be a problem with multiple concurrent execution.

The timeout when evaluating the fastboot network availability is surprising - I had tested this fairly heavily on my desk and never observed issues.

Thankfully - everything you've described looks tractable and something we can make sure gets delivered with 2.0.

tdewey-rpi, 2025-04-29:
I believe this should now be comprehensively resolved - through a combination of breaking false-sharing of assets and lockfiles, I've been provisioning multiple devices by plugging in (powered) USB hubs with multiple CM5s attached.

Let me know how you get on, @jesperbnp. This was a major effort in 2.0, and I'd like to know it worked (or even if it hasn't!)

jesperbnp, 2025-04-29:
> Let me know how you get on, [@jesperbnp](https://github.com/jesperbnp). This was a major effort in 2.0, and I'd like to know it worked (or even if it hasn't!)

We ended up going a completely different route in our production flow.:

We now program the fuses (keyhash, secureboot, disable JTAG, boot-order) with usbboot. We configure USB as secondary boot mode (EMMC is primary). Then we attach a USB SSD drive with an image that boots up, prepares the EMMC, fuses a unique key for the LUKS volume and copies our golden root image to the LUKS volume. Boot partition is written as the very last part which ensures the rpi boot directly on EMMC on next boot only if flashing of the entire image was OK.

This allows us to program multiple systemes (using multiple USB SSD disks) at once without requiring them to be online or connected to the rpi-sb-provisioner. This flow proved to be more convenient and stable for us.

I.e. rpi-sb-provisioner was used mostly as a learning platform to get secure boot etc. working.