raspberrypi/rpi-sb-provisioner #146: Upgrade fails with STORAGE_CIPHER as xchacha12 or aes-adiantum-plain64

krygas-crane, 2025-03-10:
I have been trying to provision CM4 module using Raspberry Pi 4 board running Raspberry Pi OS 64 bit. I have used the stable build of provisioner from apt install. 

Config parameters are as follows:

> CUSTOMER_KEY_FILE_PEM=/home/kamil/Desktop/private.pem
> CUSTOMER_KEY_PKCS11_NAME=
> GOLD_MASTER_OS_FILE=/home/kamil/Desktop/test-pi-gen-builds/2025-03-04-tci2_image-lite.img
> RPI_DEVICE_STORAGE_TYPE=sd
> RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64
> RPI_DEVICE_FAMILY=4
> RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/home/kamil/Desktop/config_cm4_bookworm.txt
> RPI_DEVICE_LOCK_JTAG=0
> RPI_DEVICE_EEPROM_WP_SET=0
> RPI_DEVICE_FETCH_METADATA=1
> RPI_DEVICE_RETRIEVE_KEYPAIR=/home/kamil/Desktop/keys
> DEMO_MODE_ONLY=
> RPI_SB_WORKDIR=
> RPI_SB_PROVISONER_MANUFACTURING_DB=/home/kamil/Desktop/provisioning-db.db

With this setup, update process works without any issues and everything completes correctly as expected.

I have tried to follow the guidelines of using xchacha12,aes-adiantum-plain64 as RPI_DEVICE_STORAGE_CIPHER for a non RPI5 model (in our case CM4) - however, this currently results in error at step 

> fastboot oem cryptinit "${RPI_DEVICE_STORAGE_TYPE}"p2 root "${RPI_DEVICE_STORAGE_CIPHER}"

Journalctl shows the following

[image]

When checking the log printed by the CM4 to UART, I get the following:

[image]

This error happens for both xchacha12 and aes-adiantum-plain64.

Any ideas what could be missing on my end?

jesperbnp, 2025-03-10:
One thing you need to know is that these lines does not do what you think. Not sure if that is your problem though:

```
RPI_DEVICE_LOCK_JTAG=0
RPI_DEVICE_EEPROM_WP_SET=0
```

The code in rpi-sb-provisioner.sh (`if [ -n "${RPI_DEVICE_EEPROM_WP_SET}" ]; then`) checks for a non-empty string so your '0' value results in true i.e. you are write-protecting your eeprom.

I'm using these settings (even though xchacha20 is not officially supported) which seems to work. I have all other kinds of propblems though :-)

```
RPI_DEVICE_STORAGE_TYPE=emmc
RPI_DEVICE_STORAGE_CIPHER=xchacha20,aes-adiantum-plain64
RPI_DEVICE_FAMILY=4ig
RPI_DEVICE_LOCK_JTAG=1
RPI_DEVICE_EEPROM_WP_SET=
RPI_DEVICE_FETCH_METADATA=
```

krygas-crane, 2025-03-10:
Thanks for your feedback about empty strings, I will fix those in my config :)

Also from your config I can see that the cipher should be "xchacha12,aes-adiantum-plain64"  and not "xchacha12" or "aes-adiantum-plain64"  (I've just tried building with your "xchacha20,aes-adiantum-plain64" and that passed the step where it failed before). 

Closing this one, good old "user error".

closed 2025-03-10 by krygas-crane