raspberrypi/rpi-sb-provisioner #148: Trouble on a second flash

mikepurvis, 2025-03-18:
Host is a Pi 5, target is a CM4 on a custom carrier.

I was able to successfully flash once, but I'm striking out on a second go.

To get it started at all, I have to delete the log directory for the target's serial:
```
sudo rm -rf /var/log/rpi-sb-provisioner/0ae9f858
```

On the second flash, I got stuck and had to manually reset the board at this point:
```
Mar 18 14:14:43 mikepurvis-pi rpi-sb-provisioner.sh[3151]: Boot directory '/srv/pi-sb-provisioner.Ham'
Mar 18 14:14:43 mikepurvis-pi rpi-sb-provisioner.sh[3151]: Loading: /srv/pi-sb-provisioner.Ham/bootfiles.bin
Mar 18 14:14:43 mikepurvis-pi rpi-sb-provisioner.sh[3151]: Using /srv/pi-sb-provisioner.Ham/bootfiles.bin
Mar 18 14:14:43 mikepurvis-pi rpi-sb-provisioner.sh[3151]: Waiting for BCM2835/6/7/2711/2712...
Mar 18 14:14:55 mikepurvis-pi rpi-sb-provisioner.sh[3151]: 2711/ position 00000200 size 0          <------------------ here
Mar 18 14:14:55 mikepurvis-pi rpi-sb-provisioner.sh[3151]: 2711/bootcode4.bin position 00000400 size 68696
Mar 18 14:14:55 mikepurvis-pi rpi-sb-provisioner.sh[3151]: Completed file-read 2711/bootcode4.bin in archive /srv/pi-sb-
```

Then it broke down at the fastboot reset:
```
Mar 18 14:09:55 mikepurvis-pi rpi-sb-provisioner.sh[2949]: Created image /tmp/tmp.uDIzSuw0Nk/boot.img
Mar 18 14:09:55 mikepurvis-pi rpi-sb-provisioner.sh[2984]: /tmp/tmp.uDIzSuw0Nk/boot.img: DOS/MBR boot sector, code offset 0x3c+2, OEM-ID "mkfs.fat", sectors/cluster 4, reserved sectors 4, FAT  1, root entries 256, Media descriptor 0xf8, sectors/FAT 152, sectors/track 32, heads 8, sectors 153600 (volumes > 32 MB), serial number 0x6ef5370f, label: "BOOT       ", FAT (16 bit)
Mar 18 14:09:56 mikepurvis-pi rpi-sb-provisioner.sh[2994]: 512+0 records in
Mar 18 14:09:56 mikepurvis-pi rpi-sb-provisioner.sh[2994]: 512+0 records out
Mar 18 14:09:56 mikepurvis-pi rpi-sb-provisioner.sh[2994]: 536870912 bytes (537 MB, 512 MiB) copied, 0.47897 s, 1.1 GB/s
Mar 18 14:10:04 mikepurvis-pi rpi-sb-provisioner.sh[2995]: mkfs.fat 4.2 (2021-01-31)
Mar 18 14:10:07 mikepurvis-pi rpi-sb-provisioner.sh[3010]: < waiting for any device >       <---------------------- here
Mar 18 14:12:07 mikepurvis-pi rpi-sb-provisioner.sh[2368]: "fastboot getvar version" failed, timed out.
Mar 18 14:12:07 mikepurvis-pi rpi-sb-provisioner.sh[3011]: rm: cannot remove '/tmp/tmp.uDIzSuw0Nk/rpi-boot-img-mount': Device or resource busy
Mar 18 14:12:08 mikepurvis-pi rpi-sb-provisioner.sh[3011]: rm: cannot remove '/tmp/tmp.uDIzSuw0Nk/rpi-rootfs-img-mount': Device or resource busy
```

Anyone got any ideas what could be going on here?

mikepurvis, 2025-03-18:
I see at that first failure, the green LED is flashing 3/4 which per [this page](https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes) is "Secure-boot configuration is not valid". 

And from other documentation it looks like this device may be toast as the key was programmed to one-time writeable memory and I deleted the key. Whoops!

tdewey-rpi, 2025-03-25:
> I see at that first failure, the green LED is flashing 3/4 which per [this page](https://www.raspberrypi.com/documentation/computers/configuration.html#led-warning-flash-codes) is "Secure-boot [...]

Unfortunately that's the only way I'm aware of (assuming you broadly followed instructions, and didn't find a way to overwrite all of OTP) to 'brick' a device.

As with any key material, I would suggest treating your signing key like the One Ring:
- Keep it secret
- Keep it safe

Closing, as there's no action I can take to mitigate losing the One Ring.

closed 2025-03-25 by tdewey-rpi