raspberrypi/rpi-sb-provisioner #154: Cannot use RPI_DEVICE_BOOTLOADER_CONFIG_FILE

ahnaf-chowdhury, 2025-04-07:
Hello,

I have been trying to provision a device (CM4) alongside a given file for RPI_DEVICE_BOOTLOADER_CONFIG_FILE ([config.txt](https://github.com/user-attachments/files/19631512/config.txt)). However, I'm not sure the configurations in the file are being loaded.

Image used: 2024-11-19-raspios-bookworm-arm64-lite.img

I am expecting to be able to do two things:

1. Read from the RTC using 'sudo hwclock -r' - There is a pcf8563 in the device I am using.

2. See a serial device enabled, since the config file should enable UART. I am expecting a file /dev/serial0.

I am able to do both when I flash the OS without rpi-sb-provisioner, but when I do use rpi-sb-provisioner, I am not.

Can you please look into this and advise me on anything I can do on my end to resolve it?

Thanks

tdewey-rpi, 2025-04-07:
That's the wrong file.

RPI_DEVICE_BOOTLOADER_CONFIG, as per the name, expects to be provided with a _bootloader_ configuration file. An example of such a thing is the default file we provide: https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/host-support/bootloader.config

Note that file is typically used to specify items like the HDMI delay and the boot device order.

The file you've linked to, however, is a config.txt - a file that is aimed at providing options for the OS loader.

Your config.txt should be included in your OS' boot partition, as is typical. This is explicitly called out in https://github.com/raspberrypi/rpi-sb-provisioner?tab=readme-ov-file#changing-configtxt

I'd suggest you modify your base OS image, use the default bootloader configuration file, and try again.

ahnaf-chowdhury, 2025-04-07:
Thanks for the prompt response! I will try doing that.

tdewey-rpi, 2025-04-29:
Closing as this has gone stale - which I'm going to assume means the issue was correctly diagnosed.

closed 2025-04-29 by tdewey-rpi