raspberrypi/rpi-sb-provisioner #156: Yocto .wic Image with U-boot

stu-spp, 2025-04-21:
I used a Yocto wic kickstart image with U-boot in the bootloader partition; provisioning was not successful and now my CM5 seems unrecoverable. Is there anything I can do to bring it back?

Here is the configuration I used:

```shell
pi@pi5:~ $ cat /etc/rpi-sb-provisioner/config
CUSTOMER_KEY_FILE_PEM=/home/pi/key.pem
CUSTOMER_KEY_PKCS11_NAME=
GOLD_MASTER_OS_FILE=/home/pi/core-image-base-raspberrypi5.rootfs.wic
RPI_DEVICE_STORAGE_TYPE=emmc
RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64
RPI_DEVICE_FAMILY=5
RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config
RPI_DEVICE_LOCK_JTAG=
RPI_DEVICE_EEPROM_WP_SET=
RPI_DEVICE_FETCH_METADATA=
RPI_DEVICE_RETRIEVE_KEYPAIR=
DEMO_MODE_ONLY=
RPI_SB_WORKDIR=
RPI_SB_PROVISONER_MANUFACTURING_DB=
pi@pi5:~ $
```

This is what I see when I disable emmc boot and run rpiboot:
```shell
x@xps:~$ sudo rpiusbboot
RPIBOOT: build-date 2025/03/30 pkg-version local cdf90134

Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.

Waiting for BCM2835/6/7/2711/2712...

Directory not specified - trying default /usr/share/rpiboot/mass-storage-gadget64/
Sending bootcode.bin
Successful read 4 bytes
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711/2712...

Sending bootcode.bin
Failed control transfer (-7,24)
Failed to write correct length, returned -7
Waiting for BCM2835/6/7/2711/2712...
```
... ad infinitum.

tdewey-rpi, 2025-04-22:
There's a lot to unpack in here, so I'll break my response up into chunks:

- Yocto .wic support: .wic files are not supported inside of rpi-sb-provisioner 1.x as we cannot reason about their contents - in particular, which partitions (if any) are available, how they should be allocated in terms of LUKS or similar. In rpi-sb-provisioner 2.0 you will only be able to use these with the 'naked' provisioning mode that does not perform any form of secure boot or disk encryption. This is covered in the README (_inserting a customer-supplied (**created with `pi-gen`**) operating system into an encrypted container on the storage device_)

- CM5 provisioning failure after initial attempt: Your initial attempt successfully set up secure boot, but failed to image the storage with bootable media (as it was supplied an unsupported image format). As such, you're in a half-provisioned state. Why you're failing to load the fastboot gadget isn't clear from the supplied logs - but my best theory is that after attempting to provision the .wic image, the provisioning system is in a half-aborted state.

- rpiboot repeats ad infinitum: If you've not countersigned the recovery image, I'd absolutely expect this to fail - on CM5, all bootloader payloads must be signed after enabling secure boot.

With your CM5 in this state, you must emplace a correctly-signed boot.img into a FAT32 partition on the device. rpi-sb-provisioner can guide you in how we would prepare this for a Raspberry Pi OS-derived image.

If you must use Yocto, then my ability to support you is limited by my scheduling - I only have time allocated to Raspberry Pi OS-derived images, as that's part of the explicit support closure of rpi-sb-provisioner.

stu-spp, 2025-04-22:
Yes, understood. Thank you!

I'm just trying to recover this cm5 with a normal `pi-gen` generated image.

Counter signing the recovery files does bring back rpiboot!

```shell
pi@pi5:~/usbboot/secure-boot-recovery5 $ ../tools/update-pieeprom.sh -fr -k "/home/pi/key.pem"
SIGN_RECOVERY: 1
Signing recovery.original.bin as bootcode5.bin
Signing pieeprom.original.bin as pieeprom.signed_boot.bin
```
and then
```shell
pi@pi5:~/usbboot $ sudo ./rpiboot -v -d secure-boot-recovery5/
```
seems to work. But I assume this is what rpi-sb-provisioner is already (automatically) doing when I try to reprovision with:
```shell
pi@pi5:~/usbboot $ sudo touch /var/log/rpi-sb-provisioner/exxxxxx6/special-skip-keywriter
```
I still run into that fastboot timeout:
```
Apr 21 10:58:00 pi5 rpi-sb-provisioner.sh[452710]: 536870912 bytes (537 MB, 512 MiB) copied, 0.315689 s, 1.7 GB/s
Apr 21 10:58:08 pi5 rpi-sb-provisioner.sh[452729]: mkfs.fat 4.2 (2021-01-31)
Apr 21 10:58:08 pi5 rpi-sb-provisioner.sh[453259]: < waiting for any device >
Apr 21 11:00:08 pi5 rpi-sb-provisioner.sh[442887]: "fastboot getvar version" failed, timed out.
```

tdewey-rpi, 2025-04-29:
@stu-spp I'd _strongly_ suggest you re-try with 2.0.

stu-spp, 2025-04-29:
I manually installed the 2.0 release .deb, forwarded the web ui over ssh, entered my configuration options, plugged in my target device, and:

[image]

My device list shows:

[image]

And none of the log links work.

tdewey-rpi, 2025-04-29:
Two points:

1) Now that rpi-sb-provisioner has increased visibility, you can more easily see where things go wrong
2) In this case, bootstrap has actually aborted, but if you look in the service list you may see other services that fail because of the multiple times the bootloaders bring up the USB descriptors.

In this case, it appears that bootstrap has aborted early. You can probably find more information by asking for the Service log from rpi-sb-bootstrap corresponding with what port.

One final note:

Earlier in this thread, you mentioned using the reprovision flag. These have now moved:

Old: `sudo touch /var/log/rpi-sb-provisioner/exxxxxx6/special-skip-keywriter`
New: `sudo touch /var/log/rpi-sb-provisioner/special-skip-keywriter/exxxxxx6`

Newer still: `sudo touch /var/log/rpi-sb-provisioner/special-reprovision-device/exxxxxx6`, which will fully update the bootloader of the device, too.

stu-spp, 2025-04-29:
Okay, I tried the special reprovisioning case:
```
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + printf [%s] %s\n 2025-04-29 10:27:04.081 Re-signing bootcode for special re-provisioning case
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: [2025-04-29 10:27:04.081] Re-signing bootcode for special re-provisioning case
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + rpi-sign-bootcode --debug -c 2712 -i /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin -o /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin -k /home/pi/key.pem -v 0 -n 16
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]:   OFFSET                 TYPE: [  SIZE] DESCRIPTION
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 00000000                 FILE: [     0] /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 00000000                  LEN: [     4] 0
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 00000004               KEYNUM: [     4] 16
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 00000008              VERSION: [     4] 0
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 0000010c     RSA2048 - SHA256: [   256] digest 45f6c0fe9c60a7612a9519272ae0b2d6e21053b61597f3767f3892bf59689cb5 signature 8dc2f616090578517e6b4d80e0ce05a5a6924539c9ca3fe17ee6d5b870059e3f8a1fd93f77f5bd08dee4e7e2cfe4ccac700a2398d027d6974a56fb0a5569b27daced7c9c5cf0308026c404159b698bf7ac7758c0e122cce69258ca43cf761a034e51a287c65f44d361ddfa15c2c21708dabe98f18ff47715d6fc04b6cc539675a5711e60f5f888756092ba0a2ba565161dfe94b3f9e4c45f702388fb8b9418862c41939ce8697034d422e6a99ba96231fa539972e538587261255ae347b2c6b4cb5307178de4f5f9ab471bb73da1646c0ce29997f9c0cdf6b0e05f1051daaaf8182965fa75ca403f01e251582e7b36fe980ffdbd255b737a13e8d92cda981405
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: 0000010c                  RSA: [   264] /home/pi/key.pem
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1669]: Image size 532
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + [ ! -f /etc/rpi-sb-provisioner/special-skip-eeprom/efeddc86 ]
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + timeout_fatal rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + command=rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + timeout_seconds=60
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + set +e
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + log Running command with 60-second timeout: "rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1"
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + log_file=/var/log/rpi-sb-provisioner/early//dev/bus/usb/001/004/bootstrap.log
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1674]: + date +%N
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1675]: + cut -c1-3
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1672]: + date +%Y-%m-%d %H:%M:%S.272
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + timestamp=2025-04-29 10:27:04.272
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + message=Running command with 60-second timeout: "rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1"
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1676]: + dirname /var/log/rpi-sb-provisioner/early//dev/bus/usb/001/004/bootstrap.log
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + mkdir -p /var/log/rpi-sb-provisioner/early//dev/bus/usb/001/004
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + touch /var/log/rpi-sb-provisioner/early//dev/bus/usb/001/004/bootstrap.log
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + chmod 644 /var/log/rpi-sb-provisioner/early//dev/bus/usb/001/004/bootstrap.log
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + echo [2025-04-29 10:27:04.272] Running command with 60-second timeout: "rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1"
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + printf [%s] %s\n 2025-04-29 10:27:04.272 Running command with 60-second timeout: "rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1"
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: [2025-04-29 10:27:04.272] Running command with 60-second timeout: "rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1"
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1588]: + timeout 60 rpiboot -d /home/pi/rpi-sb-workdir/secure-bootloader/ -p 1-1
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: RPIBOOT: build-date 2025/04/01 pkg-version 20250401~081331
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: Please fit the EMMC_DISABLE / nRPIBOOT jumper before connecting the power and USB cables to the target device.
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: If the device fails to connect then please see https://rpltd.co/rpiboot for debugging tips.
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: Loading: /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: Waiting for BCM2835/6/7/2711/2712...
Apr 29 10:27:04 pi5 rpi-sb-bootstrap.sh[1681]: Loading: /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
Apr 29 10:27:05 pi5 rpi-sb-bootstrap.sh[1681]: Sending bootcode.bin
Apr 29 10:27:08 pi5 rpi-sb-bootstrap.sh[1681]: Successful read 4 bytes
Apr 29 10:27:09 pi5 rpi-sb-bootstrap.sh[1681]: Waiting for BCM2835/6/7/2711/2712...
Apr 29 10:27:09 pi5 rpi-sb-bootstrap.sh[1681]: Loading: /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
Apr 29 10:27:11 pi5 rpi-sb-bootstrap.sh[1681]: Sending bootcode.bin
Apr 29 10:27:12 pi5 rpi-sb-bootstrap.sh[1681]: Failed control transfer (-7,24)
[... cut ...]
Apr 29 10:29:15 pi5 rpi-sb-bootstrap.sh[1861]: Loading: /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
Apr 29 10:29:17 pi5 rpi-sb-bootstrap.sh[1861]: Sending bootcode.bin
Apr 29 10:29:18 pi5 rpi-sb-bootstrap.sh[1861]: Failed control transfer (-7,24)
Apr 29 10:29:18 pi5 rpi-sb-bootstrap.sh[1861]: Failed to write correct length, returned -7
Apr 29 10:29:19 pi5 rpi-sb-bootstrap.sh[1861]: Waiting for BCM2835/6/7/2711/2712...
Apr 29 10:29:19 pi5 rpi-sb-bootstrap.sh[1861]: Loading: /home/pi/rpi-sb-workdir/secure-bootloader//bootcode5.bin
```

stu-spp, 2025-04-29:
I ordered a few more cm5 for testing. I think this one is beyond saving.

Does rpi-sb-provisioner support images generated from https://github.com/raspberrypi/rpi-image-gen ?

tdewey-rpi, 2025-04-29:
> I ordered a few more cm5 for testing. I think this one is beyond saving.

Understood. That's not a failure I would have expected, given the same key and an operational nRPIBOOT jumper (I have such a set-up on my desk re-flashing with special-reprovision-device every few hours).

> 
> Does rpi-sb-provisioner support images generated from https://github.com/raspberrypi/rpi-image-gen ?

Yes, provided you're using an image layout akin to one from pi-gen (read: https://github.com/raspberrypi/rpi-image-gen/blob/master/image/mbr/simple_dual/README.adoc)

stu-spp, 2025-04-30:
Okay, I just tried a brand new CM5 that I got in the mail.

I used this config:

```
CUSTOMER_KEY_FILE_PEM=/home/pi/key.pem
CUSTOMER_KEY_PKCS11_NAME=
GOLD_MASTER_OS_FILE=/home/pi/rpi-image-gen/work/cage-chromium-lite/artefacts/cage-chromium-lite.img
PROVISIONING_STYLE=secure-boot
RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config
RPI_DEVICE_EEPROM_WP_SET=
RPI_DEVICE_FAMILY=5
RPI_DEVICE_LOCK_JTAG=
RPI_DEVICE_RETRIEVE_KEYPAIR=
RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64
RPI_DEVICE_STORAGE_TYPE=emmc
RPI_SB_PROVISIONER_MANUFACTURING_DB=/srv/rpi-sb-provisioner/manufacturing.db
RPI_SB_WORKDIR=/home/pi/rpi-sb-workdir
```

[image]

And it has failed. It does not work. I'm not sure what I'm missing.

tdewey-rpi, 2025-04-30:
Looking at the state log screenshot, it appears that your new device was bootstrapped correctly, and then subsequently re-plugged. If it's a CM5, this behaviour would require /etc/rpi-sb-provisioner/special-reprovision-device/<serial> to be present before bootstrapping would again be permitted.

More curious, though, is that you don't seem to have any sign of the triage service running after bootstrap has loaded Fastboot.

Can you list the masked systemd units?

`systemctl list-unit-files --state=masked | grep rpi-`

Also, can you check the status of rpi-sb-triage@.service?

`systemctl status rpi-sb-triage@.service`

I've only seen that sort of behaviour from two paths:

1) The systemd unit isn't configured correctly. You might've got there if you used a pre-release, but I tried to mitigate that in the official release packaging. It's possible there's a bug in this migration.
2) The udev rules aren't working correctly on your provisioning server. We only officially support running on Raspberry Pi OS, though I suspect in practice other operating systems may work if you can port the dependency closure.

tdewey-rpi, 2025-05-06:
Apologies - that's the wrong command for an instance unit.

Instead, could you give me the output from:

`systemctl list-unit-files rpi-*`?

This should be a table that includes the states of both persistent and instance units.

tdewey-rpi, 2025-07-29:
@stu-spp Is this issue still relevant?

stu-spp, 2025-07-30:
I couldn't get Yocto wic kickstart images working. Despite having identical partition layout. Not sure why. Darn. I switched to `rpi-image-gen` for our projects. `rpi-image-gen` generated images work great.

Thank you, Tom!!!

closed 2025-07-30 by stu-spp