raspberrypi/rpi-sb-provisioner #160: Boot.sig file and more trouble

GerethSoesoo, 2025-05-09:
Hello:
I'm a trainee and I was given a job to make a provisoner server and provison 4 CM4 machines and I was wondering what I was doing wrong since I've been stuck on a problem for some time now.
The problem is that whenever I start the provisioning, then the bootstrapper is trying to find the boot.sig file in the directory of /srv/rpi-sb-provisioner/rpi-sb-bootstrap.XXX/bootfiles.bin. But everytime it creates that folder it doesn't add the Boot.sig file into it.

The log files also show that "Bootstrap already in progress for e25e****". And was wondering where does it save the files that say that the bootstrapper has been ran already.

I'm very new to this so this has taken lot of time.
Oh, and I have one question aswell, If the CM4 has only 16GB of storage will that be a problem.
I've added the log as a txt file under this issue
[Log.txt](https://github.com/user-attachments/files/20117387/Log.txt)

tdewey-rpi, 2025-05-09:
Thanks for the report, @GerethSoesoo 

rpi-sb-provisioner, for CM4, will not use the bootfiles.bin approach that is required on Zero2W. Instead, it should be using the boot.img/boot.sig pair.

Looking at the log, it would appear that you've managed to write the signing_key, but then failed to boot fastboot - possibly because it's not signed.

I'll try to reproduce your case today.

GerethSoesoo, 2025-05-12:
Thank you for the quick answer, since I wasn't in the office at the time I'm writing you later. 
I have a question about the "Bootstrap already in progress for e25e****". If I have an testing CM4 and I want to keep attempting to provision it, if the first time failed, do I need to delete the lock files? Cause if so I have and it still reads the message "Bootstrap already in progress for e25e****".

GerethSoesoo, 2025-05-14:
Hi again,

I’ve been continuing work on the CM4s and wanted to give an update.

After resolving the initial problem with the missing boot.sig and the provisioning system not launching fastboot properly, I’ve been trying to recover one of the CM4s that seems to be in a bad state. It’s not responding to provisioning anymore, and I suspected the SPI EEPROM or the eMMC might be at fault.

So I tried putting the CM4 into recovery mode via USB boot using rpiboot. I prepared a mass-storage gadget boot folder (msd), renamed bootcode.bin to bootcode4.bin, and included start.elf, fixup.dat, cmdline.txt with modules-load=dwc2,g_mass_storage, and a minimal config.txt. I’m running rpiboot -d on Windows.

It does load the bootcode (Sending bootcode.bin and Successful read 4 bytes), but then just sits at Second stage boot server or sometimes repeats the Sending bootcode.bin line in a loop — which looks like the Pi is rebooting or failing to continue into the MSD stage.

I confirmed the boot files are consistent and pulled from a known-good Raspberry Pi OS image. I also tried the same setup on my Raspberry Pi 5 (as a host) using the Linux rpiboot, and behavior is the same — bootcode gets sent, but the CM4 never exposes its eMMC as a USB drive.

So far this looks like either:

The CM4’s eMMC is unreadable, or

It crashes when trying to load start.elf or bring up USB MSD.

tdewey-rpi, 2025-05-19:
> Hi again,
>
> I’ve been continuing work on the CM4s and wanted to give an update.
>
> After resolving the initial problem with the missing boot.sig and the provisioning system not launching fastboot [...]

That's concerning. How did you acquire the CM4 device? Are you able to boot another Raspberry Pi device (Raspberry Pi 5, for example) over rpiboot, to confirm there isn't something wrong with the boot files?