raspberrypi/rpi-sb-provisioner #165: Provisioning failed

MarkaRagnos0815, 2025-05-14:
Hello

I tried provisioning via your tool
I followed the installation/setup step from your readme
- Setup a PI4 as server
- Setup a CM4 as target board 
- upload a image to PI4 server
- connect the CM4 board via usb (it shows up in device list, jumper to disable emmc boot)
- I changed options: set image file, select provisioning style to naked/secure boot (tried borh), added a pem file and storage type to emmc
- saved the options

following your guide I should now call rpi-sb-provisioning.sh, right?

The output looks like 
`
+ . /var/lib/rpi-sb-provisioner/manufacturing-data
+ . /var/lib/rpi-sb-provisioner/state-recording
+ DEBUG=
+ OPENSSL=openssl
+ export PROVISIONER_FINISHED=SB-PROVISIONER-FINISHED
+ export PROVISIONER_ABORTED=SB-PROVISIONER-ABORTED
+ export PROVISIONER_STARTED=SB-PROVISIONER-STARTED
+ dirname /usr/bin/rpi-sb-provisioner.sh
+ . /usr/bin/rpi-sb-common.sh
+ LOCK_BASE=/var/lock/rpi-sb-provisioner
+ LOG_BASE=/var/log/rpi-sb-provisioner
+ TEMP_BASE=/srv/rpi-sb-provisioner
+ MAX_TEMP_DIR_AGE_HOURS=24
+ read_config
+ [ -f /etc/rpi-sb-provisioner/config ]
+ . /etc/rpi-sb-provisioner/config
+ CUSTOMER_KEY_FILE_PEM=/home/treetek/privatekey.pem
+ CUSTOMER_KEY_PKCS11_NAME=
+ GOLD_MASTER_OS_FILE=/srv/rpi-sb-provisioner/images/strato_pi_max_sx.img
+ PROVISIONING_STYLE=naked
+ RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config
+ RPI_DEVICE_EEPROM_WP_SET=
+ RPI_DEVICE_FAMILY=4
+ RPI_DEVICE_LOCK_JTAG=
+ RPI_DEVICE_RETRIEVE_KEYPAIR=
+ RPI_DEVICE_STORAGE_CIPHER=aes-xts-plain64
+ RPI_DEVICE_STORAGE_TYPE=emmc
+ RPI_SB_PROVISIONER_MANUFACTURING_DB=
+ RPI_SB_WORKDIR=
+ : aes-xts-plain64
+ TMP_DIR=
+ trap cleanup INT TERM
+ check_command_exists losetup
+ command_to_test=losetup
+ command -v losetup
+ echo losetup
losetup
+ check_command_exists mknod
+ command_to_test=mknod
+ command -v mknod
+ echo mknod
mknod
+ check_command_exists lsblk
+ command_to_test=lsblk
+ command -v lsblk
+ echo lsblk
lsblk
+ check_command_exists cut
+ command_to_test=cut
+ command -v cut
+ echo cut
cut
+ check_command_exists findmnt
+ command_to_test=findmnt
+ command -v findmnt
+ echo findmnt
findmnt
+ check_command_exists openssl
+ command_to_test=openssl
+ command -v openssl
+ echo openssl
openssl
+ check_command_exists python
+ command_to_test=python
+ command -v python
+ echo python
python
+ check_python_module_exists Cryptodome
+ module_name=Cryptodome
+ python -c import Cryptodome
+ echo Cryptodome
Cryptodome
+ check_command_exists zstd
+ command_to_test=zstd
+ command -v zstd
+ echo zstd
zstd
+ check_command_exists cpio
+ command_to_test=cpio
+ command -v cpio
+ echo cpio
cpio
+ check_command_exists sed
+ command_to_test=sed
+ command -v sed
+ echo sed
sed
+ check_command_exists cp
+ command_to_test=cp
+ command -v cp
+ echo cp
cp
+ check_command_exists mount
+ command_to_test=mount
+ command -v mount
+ echo mount
mount
+ check_command_exists fastboot
+ command_to_test=fastboot
+ command -v fastboot
+ echo fastboot
fastboot
+ check_command_exists simg_dump
+ command_to_test=simg_dump
+ command -v simg_dump
+ echo simg_dump
simg_dump
+ check_command_exists blockdev
+ command_to_test=blockdev
+ command -v blockdev
+ echo blockdev
blockdev
+ check_command_exists grep
+ command_to_test=grep
+ command -v grep
+ echo grep
grep
+ check_command_exists mke2fs
+ command_to_test=mke2fs
+ command -v mke2fs
+ echo mke2fs
mke2fs
+ check_command_exists img2simg
+ command_to_test=img2simg
+ command -v img2simg
+ echo img2simg
img2simg
+ check_command_exists mkfs.fat
+ command_to_test=mkfs.fat
+ command -v mkfs.fat
+ echo mkfs.fat
mkfs.fat
+ check_command_exists truncate
+ command_to_test=truncate
+ command -v truncate
+ echo truncate
truncate
+ check_command_exists systemd-notify
+ command_to_test=systemd-notify
+ command -v systemd-notify
+ echo systemd-notify
systemd-notify
+ setup_fastboot_and_id_vars 
+ FASTBOOT_DEVICE_SPECIFIER=
+ timeout_fatal fastboot -s  getvar version
+ command=fastboot -s  getvar version
+ set +e
+ log Running command with 30-second timeout: "fastboot -s  getvar version"
+ date +%N
+ cut -c1-3
+ date +%Y-%m-%d %H:%M:%S.625
+ timestamp=2025-05-14 16:03:49.625
+ echo [2025-05-14 16:03:49.625] Running command with 30-second timeout: "fastboot -s  getvar version"
+ timeout 30 fastboot -s getvar version
< waiting for getvar >
+ command_exit_status=124
+ record_state  SB-PROVISIONER-ABORTED 
+ touch /srv/rpi-sb-provisioner/state.db
+ sqlite3 /srv/rpi-sb-provisioner/state.db -cmd PRAGMA journal_mode=WAL; -cmd PRAGMA busy_timeout=5000; CREATE TABLE IF NOT EXISTS devices(                        id              integer primary key,                    serial          TEXT                not null,           endpoint        TEXT                not null,           state           TEXT                not null,           image           TEXT                not null,           ip_address      TEXT                not null,           ts              timestamp           default current_timestamp         );
+ serial=
+ state=SB-PROVISIONER-ABORTED
+ endpoint=
+ image=
+ ip_address=
+ sqlite3 /srv/rpi-sb-provisioner/state.db PRAGMA busy_timeout=5000; INSERT INTO devices(serial, endpoint, state, image, ip_address) VALUES ('', '', 'SB-PROVISIONER-ABORTED', '', '');
+ die "fastboot -s  getvar version" FAILED: Timed out after 30 seconds (exit code 124).
+ record_state  SB-PROVISIONER-ABORTED 
+ touch /srv/rpi-sb-provisioner/state.db
+ sqlite3 /srv/rpi-sb-provisioner/state.db -cmd PRAGMA journal_mode=WAL; -cmd PRAGMA busy_timeout=5000; CREATE TABLE IF NOT EXISTS devices(                        id              integer primary key,                    serial          TEXT                not null,           endpoint        TEXT                not null,           state           TEXT                not null,           image           TEXT                not null,           ip_address      TEXT                not null,           ts              timestamp           default current_timestamp         );
+ serial=
+ state=SB-PROVISIONER-ABORTED
+ endpoint=
+ image=
+ ip_address=
+ sqlite3 /srv/rpi-sb-provisioner/state.db PRAGMA busy_timeout=5000; INSERT INTO devices(serial, endpoint, state, image, ip_address) VALUES ('', '', 'SB-PROVISIONER-ABORTED', '', '');
+ echo "fastboot -s  getvar version" FAILED: Timed out after 30 seconds (exit code 124).
"fastboot -s  getvar version" FAILED: Timed out after 30 seconds (exit code 124).
+ exit 1
`

So it seems to have a problem with the database I guess . Can you just help me to figure out whats the problem?

Thanks in advance

tdewey-rpi, 2025-05-15:
@MarkaRagnos0815 Good news: You don't need to manually call, well, anything!

After setting up the tool in the WebUI (available at localhost:3142, or with a self-signed certificate at localhost:3143), you should be able to just connect your target device in RPIBOOT mode, and it'll run. You can watch the progress of the device in the WebUI.

tdewey-rpi, 2025-05-19:
Closing as a user error - I may consider further revisions to the documentation at a later date, however I note that the current docs do not advise running the systemd service scripts directly.

closed 2025-05-19 by tdewey-rpi