raspberrypi/rpi-sb-provisioner #170: Stuck at "Cannot open file boot.sig"

octal-ip, 2025-05-25:
I've attempted to provision a new CM4 using rpi-sb-provisioner according to the documentation, however the first attempt quickly failed. Subsequent attempts are also failing indicating that boot.sig is missing, as is any recovery effort using rpiboot.

Below are the logs from the first attempt:

/var/log/rpi-sb-provisioner/early/dev/bus/usb/001/006/bootstrap.log
> [2025-05-25 13:21:09.210] ================================================================================
[2025-05-25 13:21:09.243] Starting Finding the cache directory
[2025-05-25 13:21:09.281] ================================================================================
[2025-05-25 13:21:09.319] ================================================================================
[2025-05-25 13:21:09.354] Stopping Finding the cache directory: Using specified name
[2025-05-25 13:21:09.390] ================================================================================
[2025-05-25 13:21:09.536] Creating secure bootloader for future reuse
[2025-05-25 13:21:09.582] ================================================================================
[2025-05-25 13:21:09.621] Starting Setting up the environment for a signed-boot capable device
[2025-05-25 13:21:09.661] ================================================================================
[2025-05-25 13:21:09.969] update_eeprom() src_image: "/lib/firmware/raspberrypi/bootloader-2711/default/pieeprom-2025-05-08.bin"
[2025-05-25 13:21:10.017] Signing bootloader config
[2025-05-25 13:21:11.819] Writing key and EEPROM configuration to the device
[2025-05-25 13:21:11.835] Running command with 60-second timeout: "rpiboot -d /media/usb/rpi-sb-provisioner/work/secure-bootloader/ -p 1-1.3"
[2025-05-25 13:21:24.939] "rpiboot -d /media/usb/rpi-sb-provisioner/work/secure-bootloader/ -p 1-1.3" succeeded with exit code 0.
[2025-05-25 13:21:24.955] Keywriting completed. Silently rebooting for next phase.
[2025-05-25 13:21:25.081] ================================================================================
[2025-05-25 13:21:25.096] Starting fastboot initialisation
[2025-05-25 13:21:25.111] ================================================================================
[2025-05-25 13:21:25.160] Running command with 60-second timeout: "rpiboot -v -d /media/usb/rpi-sb-provisioner/work -p 1-1.3"
[2025-05-25 13:22:25.227] ================================================================================
[2025-05-25 13:22:25.251] Starting Deleting udev lock
[2025-05-25 13:22:25.273] ================================================================================
[2025-05-25 13:22:25.322] ================================================================================
[2025-05-25 13:22:25.343] Stopping Deleting udev lock
[2025-05-25 13:22:25.365] ================================================================================
[2025-05-25 13:22:25.381] ================================================================================
[2025-05-25 13:22:25.397] Starting Deleting public key
[2025-05-25 13:22:25.413] ================================================================================
[2025-05-25 13:22:25.430] ================================================================================
[2025-05-25 13:22:25.446] Stopping Deleting public key
[2025-05-25 13:22:25.462] ================================================================================

Service Log: rpi-sb-bootstrap@dev-bus-usb-001-006.service
> 2025-05-25 13:22:25 [2025-05-25 13:22:25.462] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.446] Stopping Deleting public key
2025-05-25 13:22:25 [2025-05-25 13:22:25.430] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.413] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.397] Starting Deleting public key
2025-05-25 13:22:25 [2025-05-25 13:22:25.381] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.365] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.343] Stopping Deleting udev lock
2025-05-25 13:22:25 [2025-05-25 13:22:25.322] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.273] ================================================================================
2025-05-25 13:22:25 [2025-05-25 13:22:25.251] Starting Deleting udev lock
2025-05-25 13:22:25 [2025-05-25 13:22:25.227] ================================================================================
2025-05-25 13:22:25 "rpiboot -v -d /media/usb/rpi-sb-provisioner/work -p 1-1.3" FAILED: Timed out after 60 seconds (exit code 124).
2025-05-25 13:22:25 Cannot open file boot.sig
2025-05-25 13:22:25 libusb_bulk_transfer sent 0 bytes; returned 0
2025-05-25 13:22:25 File 2711/boot.sig not found in /media/usb/rpi-sb-provisioner/work/bootfiles.bin
2025-05-25 13:22:25  position 00160600 size 0
2025-05-25 13:22:25  position 00160400 size 0
2025-05-25 13:22:25  position 00160200 size 0
2025-05-25 13:22:25  position 00160000 size 0
2025-05-25 13:22:25  position 0015fe00 size 0
2025-05-25 13:22:25  position 0015fc00 size 0
2025-05-25 13:22:25  position 0015fa00 size 0
2025-05-25 13:22:25  position 0015f800 size 0
2025-05-25 13:22:25  position 0015f600 size 0
2025-05-25 13:22:25 2712/memsys01.bin position 00155200 size 40972
2025-05-25 13:22:25 2712/font.bin position 0014e400 size 27404
2025-05-25 13:22:25 2712/memsys00.bin position 00144000 size 40972
2025-05-25 13:22:25 2712/mcb.bin position 00142800 size 5460
2025-05-25 13:22:25 2712/memsys03.bin position 00138400 size 40972
2025-05-25 13:22:25 2712/logo.bin position 00134e00 size 12844
[... 63 lines cut ...]
2025-05-25 13:22:25  position 00160200 size 0
2025-05-25 13:22:25  position 00160000 size 0
2025-05-25 13:22:25  position 0015fe00 size 0
2025-05-25 13:22:25  position 0015fc00 size 0
2025-05-25 13:22:25  position 0015fa00 size 0

All subsequent attempts seem to fail with the same missing boot.sig issue repeating a few times. Unsurprisingly, the CM4 also won't boot in normal mode anymore.

My configuration:
>CUSTOMER_KEY_FILE_PEM=/media/usb/private-key.pem
CUSTOMER_KEY_PKCS11_NAME=
GOLD_MASTER_OS_FILE=/media/usb/pi-gen/deploy/2025-05-24-Base-v1-lite.img
PROVISIONING_STYLE=secure-boot
RPI_DEVICE_BOOTLOADER_CONFIG_FILE=/var/lib/rpi-sb-provisioner/bootloader.config
RPI_DEVICE_EEPROM_WP_SET=
RPI_DEVICE_FAMILY=4
RPI_DEVICE_LOCK_JTAG=
RPI_DEVICE_RETRIEVE_KEYPAIR=/media/usb/rpi-sb-provisioner/keypairs
RPI_DEVICE_STORAGE_CIPHER=xchacha12,aes-adiantum-plain64
RPI_DEVICE_STORAGE_TYPE=emmc
RPI_SB_PROVISIONER_MANUFACTURING_DB=/media/usb/rpi-sb-provisioner/manufacturing.db
RPI_SB_WORKDIR=/media/usb/rpi-sb-provisioner/work

My problem seesm similar to issue #160 , however the user did not post how they managed to move past the boot.sig problem.
I'd appreciate any advice on what might have gone wrong, and how I can recover the CM4 to a working state again so I can attempt another sb provisioning.

octal-ip, 2025-05-28:
Nice one, thanks @tdewey-rpi .

tdewey-rpi, 2025-05-29:
2.0.3 has been cut with that fix, and I'm going to try and get a CM4 rig together today to confirm  we're in a good place - unless someone can report earlier.

BustaC, 2025-06-23:
Here's a patch that for me fixes the boot.sig issue. It's not fully tested and is patched against 2.0.4

[rpi-sb-bootstrap.sh.patch](https://github.com/user-attachments/files/20873029/rpi-sb-bootstrap.sh.patch)

octal-ip, 2025-07-11:
A big thanks to everyone - I've successfully run rpi-sb-provisioner 2.0.5 against the previously problematic CM4 with no issues. It's now up and running with my custom image.
I really appreciate the effort on this!

tdewey-rpi, 2025-07-11:
Closing as "fixed", per OP's report.