raspberrypi/rpi-sb-provisioner #179: Triage service not starting after bootstraping

roszekF, 2025-06-06:
Hello,
I've been tying to provision a Pi5 with another Pi5 - just a basic setup with the raspios-lite generated via pi-gen and "naked" provisioning style, no luck so far.
It seems that the Pi does not boot up correctly after the bootstrap service is done and so the triage service does not start to continue the provisioning. lsusb does not show anything and the Pi5 status light is red.

Here are the logs from bootstrap service:
[rpi-sb-bootstrap.txt](https://github.com/user-attachments/files/20630635/rpi-sb-bootstrap.txt)

I have also found that I need to manually reconnect the Pi5 in RPIBOOT mode after bootstrap-firmware-updated for the bootstrap to complete, otherwise it timeouts.

roszekF, 2025-06-07:
Alright, my main issue was solved by using an active USB hub to connect the RPis. I guess my power supply is not good enough.

The manual re-connection is still needed during the bootstrap stage, otherwise it times out:
`2025-06-07 18:04:01 "rpiboot -v -d /srv/rpi-sb-provisioner/rpi-sb-bootstrap.aPH -p 3-1.1.1.1" FAILED: Timed out after 60 seconds (exit code 124).
2025-06-07 18:03:01 Waiting for BCM2835/6/7/2711/2712...`

Am I doing something wrong or is it just how it is?

przemoch-dev, 2026-01-15:
Hello @tdewey-rpi, I encounter this problem when I want to provision the OS on CM5 using IO Board. Do you have any suggestions?