raspberrypi/rpi-sb-provisioner #180: Key for partition decryption storage

snedosiekov, 2025-06-07:
Hello Tom

I have a question about place of decryption key storage on the target device, i.e. that have been provisioned. Are it stored securely to resist reverse engineering treat ? Or to provide decryption in security perimeter, i.e. our decryption key not stored in RAM or an some part of storage with read access, we need to use some external decoder device? About RAM - looks like there are PCIe vulnerability https://github.com/m1geo/Pi5_PCIe

Regards, Serhii

tdewey-rpi, 2025-06-09:
> About RAM - looks like there are PCIe vulnerability https://github.com/m1geo/Pi5_PCIe

I'm not sure how this is a vulnerability - as far as I can see, M1GEO has created a board to expose a more traditional PCIe connector than the FPC on Raspberry Pi 5? I also took a quick look through their website, didn't see anything referencing a system DRAM compromise. As ever, if you're aware of a specific security concern please follow the security policy in this repository: https://github.com/raspberrypi/rpi-sb-provisioner?tab=security-ov-file

tdewey-rpi, 2025-06-09:
> > ... revoking the development key (note that the last item, revoking the development key, is done automatically in recent firmware, and that is enforced with rpi-sb-provisioner 2.0.3+)
> 
> Could you expand on _revoking the development key_? There is an already existing development key burned into OTP? Does this sit next to the public hash of the `CUSTOMER_KEY_FILE_PEM` in OTP?

Two diagrams will help: 

2712: https://github.com/raspberrypi/usbboot/blob/master/docs/secure-boot-chain-of-trust-2712.pdf
2711: https://github.com/raspberrypi/usbboot/blob/master/docs/secure-boot-chain-of-trust-2711.pdf

The key (no pun intended) thing to note between the two diagrams is that on 2712, the second stage (post-ROM) is _countersigned_ by both Raspberry Pi and your customer signing key. This is a different behaviour to 2711, where the second stage bootloader is signed only by Raspberry Pi.

Regardless of platform, the third stage (Linux) is then signed with the customer key.

Revoking the development key is therefore exclusively relevant on 2711 (Raspberry Pi 4-family) devices - to prevent someone inserting a BETA or testing bootloader release that may not correctly enforce signed boot.

With that in mind, we changed the default behaviour of the bootloader: https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware-2711/release-notes.md#2025-05-16-2711-automatically-set-revoke_devkey-if-program_pubkey1-latest

So if you've provisioned a Raspberry Pi 4-family device with rpi-sb-provisioner 2.0.3 or later, you'll find we automatically performed that revocation. If you provision a Raspberry Pi 5-family device, it was never an issue.

timg236, 2025-06-12:
> Hi. My friend from the security research team tried to implement reading RAM via the PCIe+DMA on the RPi5. And all the memory turned out to be accessible on all Raspberry Pi distributions. I think this is a serious obstacle to using secure boot on RPi5. I don't quite understand at what stage of loading the iommu is initially initialized (at the boot stage or inside Linux). The first thing that comes to mind is to completely disable the PCIe interface. I ordered a FPGA board to experiment with the PCIe, I can tell you a little later what I got

There isn't an IOMMU for PCIe on BCM2711 or BCM2712 so this is one of the known limitations of physical attacks that can't be completely blocked.

It would be useful to know the bootloader config i.e. is NVMe boot enabled and whether the FPGA was attached to the x1 connector or if RP1 was replaced. PCIe devices on the x1 interface should not be enumerated in early boot unless NVMe boot is enabled.

tdewey-rpi, 2025-06-12:
> today I ordered CaptainDMA M.2 card. I think in about a week and a half I will send a report with logs from UART

Thanks for the update, @OnlySlon.

We'll continue investigations, and await your logs.

timg236, 2025-06-13:
> Hi all. As an idea how to detect PCIe attack attempt is on boot stage compare PCIe bus enum with etalon one and if thay have a differance - dont process future boot. With plug and play scenario - in CPIe IRQ on this event do the same and in case this is not allowed device - trunc sensative memory area and reset/halt cpu
> 
> All of this looks doible, but may be I miss something

We really need the logs and full description of the attach (via security@raspberrypi.com) to comment further. The pre-boot environemnt does not plug-and-play PCIe devices. It's more limited than that.

OnlySlon, 2025-06-13:
> Hi all. As an idea how to detect PCIe attack attempt is on boot stage compare PCIe bus enum with etalon one and if thay have a differance - dont process future boot. With plug and play scenario - in [...]

This is not a solution. There are fpga cards in the wild that can transparently pass data from a real card, but inject memory read/write commands.

OnlySlon, 2025-07-19:
Still waiting for dma-card delivery.

OnlySlon, 2025-08-06:
> > Hi all. As an idea how to detect PCIe attack attempt is on boot stage compare PCIe bus enum with etalon one and if thay have a differance - dont process future boot. With plug and play scenario - in CPIe IRQ on this event do the same and in case this is not allowed device - trunc sensative memory area and reset/halt cpu
> > All of this looks doible, but may be I miss something
> 
> We really need the logs and full description of the attach (via [security@raspberrypi.com](mailto:security@raspberrypi.com)) to comment further. The pre-boot environemnt does not plug-and-play PCIe devices. It's more limited than that.

Finally I Got DMA card, and looks like I can read RAM. I can give you remote access to my lab. I have two rpi5 with uart connection and linux utils for reading memory via pcie

OnlySlon, 2025-08-06:
config.txt
`# For more options and information see
# http://rptl.io/configtxt
# Some settings may impact device functionality. See link above for details

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Enable audio (loads snd_bcm2835)
dtparam=audio=on
#dtparam=pciex1
#dtparam=nvme
#dtparam=pciex1_gen=3

# Additional overlays and parameters are documented
# /boot/firmware/overlays/README

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Automatically load initramfs files, if found
auto_initramfs=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
disable_fw_kms_setup=1

# Run in 64-bit mode
arm_64bit=1

# Disable compensation for displays with overscan
disable_overscan=1

# Run as fast as firmware / board allows
arm_boost=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[cm5]
dtoverlay=dwc2,dr_mode=host

[all]
`

UART bootlog
`  0.73 RPi: BOOTSYS release VERSION:26826259 DATE: 2024/09/23 TIME: 14:02:56
  0.77 BOOTMODE: 0x06 partition 0 build-ts BUILD_TIMESTAMP=1727096576 serial bb75da5e boardrev b04170 stc 777532
  0.87 AON_RESET: 00000003 PM_RSTS 00000020
  0.95 RP1_BOOT chip ID: 0x20001927
  0.97 PM_RSTS: 0x00000020
  0.97 part 00000000 reset_info 00000000
  0.01 PMIC reset-event 00000000 rtc 6893cfe9 alarm 00000000 enabled 0
  0.07 uSD voltage 3.3V
  0.26 Initialising SDRAM rank 2 total-size: 64 Gbit 4267 (0x1c 0x00)
  0.30 DDR 4267 1 1 64 152
  2.24 OTP boardrev b04170 bootrom a a
  2.25 Customer key hash 0000000000000000000000000000000000000000000000000000000000000000
[... 105 lines cut ...]
`

Looks like memory dump
`
root@raspberrypi:/home/user/WORK/BIN# strings /dev/shm/mem.bin |grep kernel
__kernel_clock_gettime
__kernel_gettimeofday
__kernel_clock_getres
__kernel_rt_sigreturn
__kernel_getrandom
__do_kernel_fault
mmio_exit_kernel
         e.g. mount -t tracefs [-o [gid=<gid>]] nodev /sys/kernel/tracing
  trace_marker          - Writes into this file writes into the kernel buffer
  trace_marker_raw              - Writes into this file writes binary data into the kernel buffer
  kprobe_events         - Create/append/remove/show the kernel dynamic events
             kernel return probes support: $retval, $arg<N>, $comm
kernel built without CONFIG_PAGE_POOL_STATS
cannot modify bits past kernel bitset size
IPv6 support not enabled in kernel
4WARNING: 'bootconfig' found on the kernel command line but CONFIG_BOOT_CONFIG is not set.
Booting kernel
5Unknown kernel command line parameters "%s", will be passed to user space.
unused kernel image (initmem)
`

timg236, 2025-08-12:
I don't believe dtparam=pcie=off will do anything if the HAT has already probed for PCIe. If you want to disable PCIe x1 in your setup then you could disable that in the DTB file itself.