raspberrypi/rpi-sb-provisioner #186: Question: Is there a way to provision Raspberry Pi Zero 2 for secure boot?

oliora, 2025-07-13:
I wonder, is there a way to provision Raspberry Pi Zero 2 for secure boot? Is it even possible in principle?

tdewey-rpi, 2025-07-21:
Unfortunately, this isn't possible today. The secure boot mechanisms used in Raspberry Pi 4-family devices and Raspberry Pi 5-family devices provide you with a durable (and critically: upgradable) mechanism to deploy kernels & OSes.

Choices made in the development of Raspberry Pi Zero 2-family devices, however, make implementing an identical scheme unviable.

If you have a specific application request, of course, the best path forward is to talk to applications@raspberrypi.com.

oliora, 2025-07-21:
Thank you for the information

closed 2025-07-29 by tdewey-rpi