raspberrypi/rpi-sb-provisioner #202: Clarification on whether rpi-image-gen is a supported source of images

jsullivan-mitre, 2025-09-16:
First off, thank you all for the great support on this and other Pi software projects. Reading through the issues and PRs here, you've been absolutely stellar in responding to people and addressing what you can.

I just wanted to ask a clarifying question about sources of images to be used with the provisioning process. pi-gen is mentioned a few times (e.g. [here](https://github.com/raspberrypi/rpi-sb-provisioner/blob/main/docs/config_vars.adoc#gold_master_os_file)) as the preferred source, but it seems like some people have had success using rpi-image-gen to build a base image for their particular application.

Is that a supported way to build images for this? Hopefully, I didn't miss a doc somewhere saying yes/no...

tdewey-rpi, 2025-09-16:
Thanks for the kind words, @jsullivan-mitre 

Rpi-image-gen is indeed the best source for images to feed to rpi-sb-provisioner, though the provisioner is only lightly opinionated - it has a handful of assumptions that basically come down to "Does it look like Debian?", but there's no explicit allowlist or similar.

Expect to see more details on how to combine the two tools in the near future - initial works have been implemented in our fastboot gadget, but I've not yet tested these in anger.

jsullivan-mitre, 2025-09-16:
Of course, @tdewey-rpi :)

Thanks for confirming that rpi-image-gen is a good source. Should be able to get that working, just want to stick as close as possible to what you all use to avoid gremlins.

closed 2025-09-16 by jsullivan-mitre