raspberrypi/rpi-sb-provisioner #203: Cannot flash CM5 module

garylee87, 2025-09-30:
Thanks for sharing good tool.

I'm trying to utilize rpi-sb-provisioner onto the production line.
Hence, I've first configured system and tools but i couldn't flash it success. Here i share the setup and log. Would you help me out what can I try?

## Hardware 

[image]

- Raspberry Pi5 (Provision host with 128G Samsung SD Card)
- CM5 IO Board
- CM5 Module (8G, 64G, no wireless option)
- Raspberry 27W 5V Power Supply 
- USB A to C 3.0 Cable
- Jumpwire (like capture above)

## Host Setup
- I have ssh server configured and apt updated, rpi-sb-provisioner installed on to fresh raspberry pi 5. nothing more have installed
- I access remote from "Mac book (guess doesn't really matter)" and accessed

```
$ ssh -L 3142:localhost:3142 <user>@<ip>
```

## Flash Failure
### Options

[image]

[image]

 - Target device is CM5 ,so I choose 
   - RPI_DEVICE_FAMLIY=5 
   - RPI_DEVICE_STORAGE_TYPE=emmc
 - I just want to basic rpi os onto the device for the first trial
   - PROVISIONG_STYPE=naked

### Images

[image]
 - Image for CM5 from the rpi official website

### Device List
 - Host recognize CM5 devices
[image]

### Services
- Once i plug USB-C cable to CM5_IO_Board service is activated but soon failure
- Here i attatch the logs 
   -  rpi-sb-bootstrap@dev-bus-usb-001-007
       - [dev-bus-usb-001-007.log](https://github.com/user-attachments/files/22614214/dev-bus-usb-001-007.log)
 - rpi-sb-bootstrap@dev-bus-usb-001-008
       - [dev-bus-usb-001-008.log](https://github.com/user-attachments/files/22614222/dev-bus-usb-001-008.log)

### Trial
 - I've tried other component with identical settings but the others also failed.
 - I've tried to flush cache and delete lock files as guided in readme.md but still have symptom

tdewey-rpi, 2025-10-06:
Thanks for the extensive report, @garylee87 

I'll try to reproduce this later today.

waiweng83, 2025-10-07:
I'm having exactly the same problem too. Is it because the rpi-sb-provisioner is only compatible with the image generated with either pi-gen or rpi-image-gen?

> Thanks for the extensive report, [@garylee87](https://github.com/garylee87)
> 
> I'll try to reproduce this later today.

stu-spp, 2025-10-07:
I note that when we were bringing up the manufacturing line for our CM5s, we couldn't get the `naked` `PROVISIONING_STYLE` flag to work. We had to go straight to `secureboot` `PROVISIONING_STYLE` and then CM5 devices started to be able to be provisioned successfully. I think this was on `v2.0.4`.

MyAccess2021, 2025-10-17:
If I have 10 CM5 boards without eMMC, I want a GUI or CLI tool that allows me to boot all of them from a single Pi. I need to flash them over the network onto NVMe drives or SD cards. The process should let me note down the MAC addresses and serial numbers of the CM5 boards and boot them with my chosen Pi-gen OS using either a CLI or web-based GUI.

tdewey-rpi, 2025-10-17:
> If I have 10 CM5 boards without eMMC, I want a GUI or CLI tool that allows me to boot all of them from a single Pi. I need to flash them over the network onto NVMe drives or SD cards. The process [...]

@MyAccess2021 

rpi-sb-provisioner will allow you to set up secure boot on all 10, provided you connect them to the provisioning server with the requisite storage attached. You would not, however, flash them over the network - secure boot cannot be enabled in that way.

I suspect your questions are better suited to the Raspberry Pi Forums, however - as they are generic workflow questions, rather than specific queries about rpi-sb-provisioner.

s-jhanvi, 2025-11-25:
@tdewey-rpi 
Thanks! The 2.1.1 version with Bookworm works for me.